westlicht/performer #2: grades A,B

mxmxmx, 2019-03-06:
hi 

any objections to (eventually) adding or pulling [16 bit support](https://github.com/mxmxmx/performer/commit/98ecfa2810e575d45b9b40ec7bff94fcfdeafe34) into your main repo? i'm not 100% but it looks as if that's all there is to it (minus the OFFSET_0995 switch of course, which obviously is tangential / irrelevant).

westlicht, 2019-03-14:
First, I don't really like adding compile switches. For that reason I added the hardware configuration section in flash which allows to reverse the encoder for example. This way, I can distribute a single firmware file supporting different hardware configurations. So if I want to support grade A/B of the DAC, I think that would have to go into the hardware config as well.

Second, even if these are 16bit DACs, they don't provide that accuracy on the outputs according to specs. With the grade C/D I actually throw away the MSB because I cannot drive full range with 3.3V VDD. This was a tradeoff, but I went for the better lower drift reference. Anyway, in terms of precision I don't think extending the calibration to full 16 bits is necessary, as they won't have that accuracy anyway. With the current implementation you can calibrate to within 1-2mV, which is more than good enough. So the only thing that would have to change switching to the grade A/B is a scaling factor of two if I'm correct.

mxmxmx, 2019-03-14:
i didn't want to start an argument about specs, people might use grade A/B for any number of reasons, like they already had them in their parts box, for example. the question really was just this: "any objections to (eventually)" adding support; if not, that's fine, too.

closed 2019-03-14 by mxmxmx

westlicht, 2019-03-14:
I think I might add support for grades A/B if the only change is the factor of two scaling. But I would do it using the hardware config so that people can set it once and not have to worry about it when upgrading the firmware.

mxmxmx, 2019-03-16:
as you prefer. the number of people ending up with grade A/B is probably going to be negligible though, so compile switch almost sounds like less of a hassle. either way, assuming things stay this way (all internal processing in "volts"), yes, the couple of changes in Calibration.h above seem to do the trick (works for calibration, regular and curve tracks); so for the time being, grade A/B support is little more than a convenience thing.