westlicht/performer #3: Slide Behaviour MIDI Output

toneburst, 2019-03-19:
What's the current behaviour for slides with MIDI output? It doesn't currently seem to have any effect.

Suggested Behaviour Options:

- Overlap incoming note and previous step note. This would work with MIDI synthesisers with a "mono legato" mode or similar.
- Transmit standard MIDI CC for Portamento (CC65), and possibly also CC5 (Portamento Time)

Each method has advantages and disadvantages.

Negatives for Method 1:
- Gate-length of previous step would be ignored.
- Previous triggered note may not be on previous step, so last note may have to be extended over several steps, which might be confusing.

Negatives for Method 2:
- Envelopes on receiving MIDI device will be re-triggered with slid notes, so not possible to emulate classic "acid-style" note-slides.

Maybe there could be an option on the MIDI output page to choose one of the above slide behaviours.

westlicht, 2019-03-19:
I did think about slides over MIDI, but as you mentioned yourself, there is no straight-forward way to implement it. I think a third option would be to have overlapping notes when the gate length of a step is at full length. This would be an easy fix and at least allow for some slide behaviour, even if it does not depend in any way of the slide layer of the sequence.

toneburst, 2019-03-20:
That sounds like a good option. How about also toggling CC65 on sequence slide steps. That way, you can have portamento on staccato notes, and acid-style slides with overlapping notes.

toneburst, 2019-03-20:
Oh, you’ve fixed it!

Will you roll a release binary, or will I have to compile from source to test?

westlicht, 2019-03-20:
The fix is on a branch currently and will be merged soon. Will be part of the next release, but I might also fix a few other things for the next release. So if you can't wait feel free to build it yourself.

toneburst, 2019-03-20:
Good to know. I can wait :)
Looking forward to trying it out when it's released, though.

closed 2019-03-22 by westlicht

toneburst, 2019-03-23:
Works really well! Great work, Simon.

commit f43442a7 (2019-03-22): implemented legato notes and slides for midi output

- legato notes are generated when gate length is set to maximum (note on
is sent before note off event)
- slides are generated by sending control change number 65

fixes #3