westlicht/performer #4: Glitchy Encoder

toneburst, 2019-03-20:
I'm finding the encoder glitchy. If turned too fast, it can reverse direction.

I'm using the specified encoder part.

I thought initially it was a hardware issue with my build (I've experienced similar behaviour with other builds, when I've forgotten to solder a capacitor related to the encoder), but a more experienced DIY builder friend noticed the same behaviour with their build.

guytaylor, 2019-03-20:
Im also getting the same behavior...

westlicht, 2019-03-20:
I actually noticed that behaviour myself a few times, but it rarely bothered me. There are no caps to filter the encoder outputs but there is software debouncing. However, I guess the scan frequency might be too low for certain edge cases. I will investigate.

toneburst, 2019-03-20:
My impression is that it’s worse when scrolling through lists of items, then when adjusting values.

When calibrating the CV outputs last night, I resorted to using the Next/Previous buttons to move through the list of outputs because using the encoder was unreliable.

However, I don’t recall noticing any issues with the encoder while adjusting the offset values, even when rotating the encoder quite rapidly.

closed 2019-03-22 by westlicht

toneburst, 2019-03-23:
I'm afraid the encoder is still glitchy with 0.1.6. My immediate impression is that it's worse than before, in fact.

As well as reversing direction, now, the selection also sometimes continues to move after I've stopped rotating the encoder.

I will try and shoot a video showing the behaviour this evening.

toneburst, 2019-03-23:
I should add this is while scrolling through menus. I haven't tested setting individual values with the encoder.

Could it be that the encoder scan-rate is lower when scrolling menus than when setting values?

westlicht, 2019-03-23:
@toneburst that's odd. Also, you should use 0.1.7 which contains the fix. 0.1.6 did not change anything regarding the encoder. And no, there is no change in scan rate. For me, 0.1.7 works flawlessly now.

toneburst, 2019-03-23:
I'm really sorry, you're right. I downloaded the 0.1.6 version, thinking that was the most recent, because I hadn't refreshed the Releases page.

Very stupid, I do apologise.

The encoder does seem much better now, thank you. I'm going out now, but will test more thoroughly when I get home this evening.

Really looking forward to testing the MIDI slide behaviour with my MB33 303 clone.

westlicht, 2019-03-23:
@toneburst no problem, let me know about your findings. Always grateful for feedback!

gsmc-alx, 2019-03-23:
It's definitely better, but still glitchy.

westlicht, 2019-03-23:
I guess I need a video. Not glitchy at all for me.

toneburst, 2019-03-23:
Above comment was me, signed in to my other GitHub account (in case you didn't realise.

I will try and made a video soon.

commit 8d1f4874 (2019-03-22): fixed glitchy encoder

fixes #4