westlicht/performer #12: Finish generator implementation and documentation

westlicht, 2019-03-23:


UncleRoel, 2019-05-28:
I don't know the generator code, but while discussing with a friend last week he mentioned a nice idea: it wound be great to have the option to mirror (up/down) and reverse(left/right) sequences, or for example double the note distances, scramble all notes, etc.

westlicht, 2019-05-29:
I have a few ideas for additional generators. But I'm also still looking for better accessibility of this feature. I'm not super happy with the current implementation but still struggle to come up with something neater.

UncleRoel, 2019-05-29:
Perhaps -depending on how many there will be- you could add pages to the context menu, with F5 advancing to the next page. This way it's a fast interface that doesn't go to a different menu style.

UncleRoel, 2019-06-01:
Inserting and deleting steps in a sequence would also be a neat feature to have in the list! Great for varying shifting patterns.

madeinspace, 2019-07-23:
An lfo generator via a curve track would be pretty handy:  |/\|/\|/\|/\

james-everitt, 2019-07-26:
Feature add for the Euclidean Gen:
Would be nice to be able to define the OFF and ON values for the selected page with F4 and F5. That would make it useful for things like octave switch or on the retrig page being able to set 'ON' to 2 retrigs/one octave up rather than always 100% of the parameter value

Also random generator based on Perlin 2D noise would be a cool generator. 8x8 that gets unwrapped to 64 steps

toneburst, 2019-08-16:
> Also random generator based on Perlin 2D noise would be a cool generator. 8x8 that gets unwrapped to 64 steps

Haven't checked the code, but I suspect the current random gen. probably IS based on Perlin noise, or something conceptually very similar.

jstefani, 2019-10-08:
I've been writing generators and the implementation seems good enough to me. My only question is if there is a reason that functions such as 

GeneratorPage::drawEuclideanGenerator

and

GeneratorPage::drawRandomGenerator

are inside ui/pages/GeneratorPage.cpp rather than inside their own classes and then called with a more generic drawSelf() function or similar in GeneratorPage.cpp

With the current implementation, GeneratorPage.cpp has to be modified for every new generator added. Not really a problem - just a comment.

westlicht, 2019-10-08:
@jstefani yes, the current implementation is bad in terms of separation. The thing is that I never got around to actually finish the generators because I've never fully settled on a concept. I have a few ideas in which direction to take them, so the current state is more of a work in progress / hack. That's also why I have not documented the feature yet, as functionality might change completely once I settle on a design.

jstefani, 2019-10-09:
I've considered using shift + track to open the context menu that the track perhaps?

captainhooke88, 2020-05-26:
I really love the generator functionality and use it a lot as is.

Would be great to have a way to affect multiple parameters at once. For instance Gate & Note.

Anyway already powerful tool, wondering what it will finally look like once you decide on the final concept.

vdgno, 2020-07-07:
I have ver 0.1.35. when entering random or euclidian generator from step screen, all sequencer buttons stop to response except function buttons. Definitely a bug here.

gitmusik, 2020-07-19:
I agree that the separation thing slows one down a little but I absolutely LOVE the Euclidean/Generator functions. I use them exhaustively and find them superior in every way to anything else on any other sequencers to my knowledge.

westlicht, 2020-07-20:
I should finally put in some work in generators. It's a great feature and I have some more ideas for other algorithms. But the UX could be better, I know that.

modularev, 2021-04-13:
@westlicht was there a change which broke the function keys inside the `EUCLIDEAN ` and `RANDOM ` mode? `INIT` `REVERT` and `COMMIT` work. Also `EUCLIDEAN\STEPS` and `RANDOM\SEED` do, but none of the other options)