westlicht/performer #14: Implement micro timing for note tracks

westlicht, 2019-03-23:
Allow shifting steps left/right in time. This is a little tricky to implement as we now have to do look ahead when playing back the sequences. This would also allow unquantized live recording.

toneburst, 2019-03-23:
YES! I was going to suggest this myself.

Maybe just allow shifting forward/back by 1/192 ticks.

toneburst, 2019-04-02:
Would capturing unquantised live recording mean that you could record multiple notes within a single step?

If not, notes would potentially not be recorded if more than one notes occurred before the next step.

westlicht, 2019-05-06:
Implementing negative step offset is a very hard problem as it requires to look into the future for scheduling steps with negative offset. This comes with some tradeoffs with regard to how reactive the sequencer can be with regards to changing things like first/last step without introducing artefacts. A first implementation will only allow for positive gate offsets, basically delaying notes. This already opens up a lot of possibilities for sequencing and is a worthy feature to have. I've implemented the data model such that negative offset can be implemented in the future. The first implementation will probably also not feature _unquantized_ recording.

closed 2019-05-06 by westlicht