westlicht/performer #16: Allow curves to span multiple steps

guytaylor, 2019-03-25:
Currently a curve is always one step in duration.
It would be useful to allow curves to span multiple steps i.e. Have a duration.
Perhaps holding both the start and the end step simultaneously whilst dialing in the curve could set a duration greater than one?

toneburst, 2019-03-25:
This is a great idea, but potentially tricky to implement, UI-wise.

What happens to the curve setting for the next step(s), when the preceding step is extended "over" it? Is that data kept, so that, if  the extended step is set back to a duration of 1, the curve setting for previously overlapped steps are restored?

westlicht, 2019-03-25:
I would rather implement duration as a layer. If a step gets longer, it will still show the same on the UI but just take longer to play back. The visualization will not be as nice but it will make much more sense in terms of the data model.

toneburst, 2019-03-25:
It would be nicer to have the visualisation show the changed length, I think.

It would be quite confusing if it didn’t, I think, in my opinion, as it wouldn’t be clear which curve steps/segments would actually play, and which wouldn’t.

I guess, as a compromise, you could “grey-out” steps/segments which which become inactive because previous steps that have been extended overlap over them.

How would different playback modes work with extended Curve segments? I’m thinking particularly of Random playback. Would the playgead only ever jump to the start of an extended segment?

westlicht, 2019-03-25:
No, all steps would still play. The ones that are "stretched" by making them longer in length would just play for longer. Currently, when you playback the sequence you can see the cursor moving through the sequence. It would just move slower on stretched steps. No steps are skipped.

toneburst, 2019-03-25:
Ah, I see. So would the sequence end step shift, to maintain the same overall length?

westlicht, 2019-03-25:
The sequence length would change yes, which could be counteracted by setting first/last step.

toneburst, 2019-03-26:
Sequence length is limited to 64 steps though, so some steps would never play, if any steps stretched over more than one step.

westlicht, 2019-03-26:
No, all steps would play, but the sequence will run for longer than a regular 64 step sequence.

ob1techno, 2019-03-26:
That would make it potentially incredibly useful for long automations!

toneburst, 2019-03-26:
That's true. Though quite hard to edit very long automation sequences in detail without the ability to scrub through them etc.

westlicht, 2019-03-31:
Started implementing step length on curve sequences. However, I've found a few tricky bits:
- `Aligned` play mode is a bit tricky to implement, as figuring out which step/substep is active becomes rather cumbersome
- Track linking is a bit tricky too, as it's not entirly clear how to handle steps that are longer
- I would also probably have to implement note repeats on the note sequence to make it similar in behaviour as the curve sequence, especially to have a correspondence between the sequence types for track linking

All in all this needs a bit more thought than I initially anticipated.

toneburst, 2019-03-31:
I have to say, I think it would be way easier to have an equivalent to tied notes for curves- ie extended segments simply overwrite subsequent steps, without extending the sequence length by pushing subsequent steps steps forward in time.

This would seem much more intuitive to me, as this is how tied notes work in a Note track, and it would seem confusing if extending steps works differently in a Curve track.

toneburst, 2019-03-31:
I realise this approach would come with other issues, but I think it makes more sense.

Just my take on things, though.

stansanders, 2019-06-15:
what if each step gained a crop left / crop right variable, so that rather than each step running a complete curve shape, it could run the same curve, but starting and ending at points that are not the end of the curve shape.  this would be useful for a couple things.  in the case of letting a curve span multiple steps, you could "expand" a curve in the UI, which would manifest by setting the subsequent steps to be fractions of the curve being expanded over them.  so if you expanded a log-up curve over 4 steps, step 1 would crop to 0-0.25 of log-up, step 2 to 0.25-0.5 of log-up, etc.  in a simple implementation this would be a one-way edit, but still useful i think.

this would also be useful in the sense that by allowing a step to index into a curve shape, you could form new shapes that aren't possible with the current way each step starts and finishes a shape with certainty.

westlicht, 2019-07-24:
After some more thoughts on this while implementing the other curve track improvements I'm a bit torn on this one.

To start of, the curve _tie_ is not a good solution. The main reason for this is that in contrast to note tracks, when curve steps are tied, their parameters (min, max etc.) get controlled from the first step in the tie sequence. This means that editing these steps should basically edit the first step of the tie sequence, which is odd. It could be handled somewhat cleanly on the edit page, but it would also require to handle this special case in the launchpad controller. This is something that is different from the existing behaviour and that's why I don't feel it's a good solution.

The idea of adding _crop_ values is better, as it would make each step still be an individual entity, completely editable. However, this will be more useful for getting "new shapes" out of the existing ones, although I feel this is not a huge advantage given the mental overhead it brings to the table. Also, to setup longer steps, this is very tedious unless I add some "macro" functionality to do this automatically. Again, this is a concept not found in any other place in the sequencer and thus I don't like the idea too much.

To conclude, I haven't found a good solution that does not bring some weirdness with it and for that reason I'm most probably opting for not supporting this. If there are some other ideas please feel free to comment. I want to keep complexity down as much as possible as I think the current system is already quite saturated in that regard.

closed 2019-07-24 by westlicht

toneburst, 2019-07-24:
I've not tried it, but since it's possible to automate track divisor from the CV output of another track using routing, so it is theoretically possible to change the length of each step in a curve sequence, as it stands (albeit by sacrificing another track to provide the automation).

jackpf, 2022-08-07:
Not sure if it totally solves your request, but I've implemented this in my own build here in case you find it useful: https://github.com/jackpf/performer/pull/8