westlicht/performer #19: Set Full-Range of Min/Max Step Values In Curve Track from Launchpad

toneburst, 2019-03-27:
Currently, the Min and Max values can only be set for steps in Curve mode in 0>63 range from an attached Launchpad, where the true range of these values is 0>255.

This limits the usefulness of Launchpad integration when editing Curve tracks (though segment curve-selection is excellent!).

possible solution:

Remap LP values to 0>255 range, so each pad represents 4 values.

closed 2019-03-29 by westlicht

toneburst, 2019-03-30:
I had thought that using the full 64-step range available would work best here, but having 8 steps is fine, I think, in practice, given you can fine-tune using the Performer UI itself, later, if required.

Great work!

westlicht, 2019-03-30:
It can be tweaked easily (one number in code), but I prefer having easy access to all values without navigating if possible. And for min/max I think it gives enough precision with 8 values.

toneburst, 2019-03-30:
Works well as it is, I think.

commit 9a0f18c3 (2019-03-29): implemented remapping of curve min/max on launchpad

fixes #19