westlicht/performer #20: Clarify launchpad model compatibility in documentation

stansanders, 2019-03-28:
Most references simply say "Launchpad" without specifying that Launchpad (NOVLPD01 / 0x000e)  will not work.  Should specify "Launchpad Mini" or others as compatibility is confirmed.

stansanders, 2019-03-28:
Also clarification on whether it's Launchpad Mini II or Launchpad Mini, or if both are supported.  Searches for the novation vendor id (0x1235) on https://devicehunt.com/search/type/usb/vendor/1235/device/any and https://www.the-sz.com/products/usbid/index.php?v=1235&p=&n= don't show devices for id 0x0035 or 0x0036.  For users purchasing a launchpad to use with their sequencer this information would be very helpful to have.

toneburst, 2019-03-28:
I think LP Mini and Mini Mk.2 are functionally identical, so both should work, assuming both are recognised by their IDs.

closed 2019-03-29 by westlicht

toneburst, 2019-03-30:
Ah, now I know how to get into Pattern mode. I had discovered button 8 can be used to mute tracks, but not that it also had other functions.

Great work, Simon!