westlicht/performer #21: Ability to capture randomised patterns

toneburst, 2019-03-30:
Some kind of ability to capture step-playback order of randomised, or random-walked steps of sequence, either to a new pattern, or to some kind of 'variation snapshot', to be stored with the current pattern.

toneburst, 2019-03-30:
Or perhaps into some kind of single temporary 'repeat buffer', allowing looping of randomised step sequence.

toneburst, 2019-03-30:
Alternatively, some kind of pattern variation generator, that would destructively shuffle/re-arrange steps using the same realtime Random and Random Walk algorithms as those used in realtime, but in this case to produce randomised, but looping variations on a pattern.

westlicht, 2019-03-30:
That's all part of "generators", have you used the existing ones (non-documented)? I have a shuffle-generator in a branch somewhere. It's one area I want to clean up soon.

toneburst, 2019-03-30:
Ah, I was wondering if Generators were the right place to put this, but though since it's operating on existing patterns, something like "Operators" might be a better category name.

I've tried the Random and Euclidean generators, and they're great!

toneburst, 2019-03-30:
Do you think there's still mileage in being able to capture shuffled patterns in realtime, though?

westlicht, 2019-03-30:
> Do you think there's still mileage in being able to capture shuffled patterns in realtime, though?

I think this will be covered by generators. The idea there is to allow for realtime modifying an existing pattern (on a buffer) and at the being able to commit or reject the changes. The UI is a work in progress, that's why it is not documented yet. There are also other generators I want to implement.

toneburst, 2019-03-30:
That’s cool, then.
Looking forward to seein more generator.

Turing Machine/shift-register?

I think there’s mileage in a shift-register Track Mode, too. I have some ideas for that. Should I open another Enhancement for them, or do you not think adding additional modes is a good idea?

westlicht, 2019-03-30:
> Turing Machine/shift-register?

I have thought about this before but it seems to be a bit of a rabbit hole. You quickly end up with something like O_C, which is great and already exists. I don't think I want to go there. I'm experimenting with a list track mode that shares some similarities with ER-101. But it allows for things like defining chord progressions, basically feeding scales that can then be used on other sequences. But it's not in a state yet that I'm happy to release.

toneburst, 2019-03-30:
Fair enough re. Turing machine.

I think it would be quite different from the o_C, though. The ability to directly set bits from step gate on/off would make it cool, I think. Plus, you could use the existing step probability layer to add variation, and add an option to allow the pattern to evolve or vary randomly, but not evolve.

I think these features would be enough to distinguish it from other shift-register-based pattern-generators.

I don’t fully understand your idea there, but sounds intriguing.

toneburst, 2019-03-30:
Ah, I think I misunderstood what you meant when you said you were wary of it becoming like the o_C.

I don't think that's a massive risk, actually. The Performer hardware is much less generic than the Ornament & Crime, so I think it's going to remain much more focussed on sequencing, of necessity.

westlicht, 2019-03-30:
My point is that there already is a lot of functionality in the sequencer. I'd rather focus on refining what is there instead of adding new orthogonal features. Especially in terms of functionality that is already well covered in other modules. Turing machine and other procedural sequencers are already available in many forms. But being open source, people can of course implement their own features or completely rewrite the firmware if they wish to.

toneburst, 2019-03-30:
Understood. And you’re right, it does already have a lot of functionality.

I may have a got at adding my own bits and pieces at a later date. It’s already a lot of fun to use though, so may never get around to it :)

closed 2019-05-13 by westlicht