westlicht/performer #22: Akai APC Mini Support

toneburst, 2019-03-31:
Add support for Akai APC Mini pad controller.

Things to consider:

“Function” buttons single-colour
Extra non-illuminated button could be used instead of Launchpad button 8.

Round buttons under and to right of 64-pad matrix are single-colour (red/green).

9x sliders. Maybe 9th slider could be used as value slider, and other sliders could be assignable on Routing page.

toneburst, 2019-03-31:
I just bought an APC Mini on eBay.

I’m happy to lend it to you for a while, if that will help, Simon.

toneburst, 2019-04-03:
My APC Mini arrived this morning :)

westlicht, 2019-04-03:
I did look into the MIDI implementation and it should be fairly straight forward to support it. But it's not high on my priority list right now. That said, you can send it to me, but I have no definite timeline for working on it. An alternative is that you just test a development branch, because I should be able to implement it without having access to the hardware. One thing I would need to know is the USB pid/vid. If you can get me that info it is a good start.

toneburst, 2019-04-03:
I can find out the hardware IDs.
There's a list of supported CC and note numbers on the Performer build thread on Muffwigglers.

I have some coding experience, so I might fork the project, and have a go at adding support for the APC pad-matrix.

One thing that would be really cool, though, I think, is if the sliders could be left unassigned for editing, and exposed as routing sources. I have a feeling that would be beyond my programming skills.

westlicht, 2019-04-03:
> One thing that would be really cool, though, I think, is if the sliders could be left unassigned for editing, and exposed as routing sources. I have a feeling that would be beyond my programming skills.

Yes, that would require a few changes. But the basic functionality to only consume messages that are required for the control surface is in place.

toneburst, 2019-04-03:
> Yes, that would require a few changes. But the basic functionality to only consume messages that are required for the control surface is in place.

That's what I thought.

Maybe if I make a start on the pad-matrix, that will save you some time, anyway.

toneburst, 2019-04-03:
Maybe the APC sliders could appear as routing sources on the Routing page, or, if not routed, could be used for editing.

It seems a shame not to be able to route them, but others might think they're better used for pattern editing.

toneburst, 2019-04-03:
> But it's not high on my priority list right now.

No problem. I'll stop nagging you about it, then ;)

> An alternative is that you just test a development branch, because I should be able to implement it without having access to the hardware.

Cool, happy to do that. I will be away on holiday next week, however, so I won't be able to test anything during that time.

> One thing I would need to know is the USB pid/vid. If you can get me that info it is a good start.

I'll see if I can get that right now, in fact.

toneburst, 2019-04-03:
```
Product ID:						0x0028
Vendor ID:						0x09e8  (AKAI  professional M.I. Corp.)
Version:						1.00
Serial Number:					Ver00.10
Speed:							Up to 12 Mb/sec
Manufacturer:					AKAI PROFESSIONAL,LP
Location ID:					0xfa120000 / 5
Current Available (mA):			500
Current Required (mA):			100
Extra Operating Current (mA):	0
```

toneburst, 2019-04-04:
Incidentally, I just read the comments re. USB power issues, which reminded me that I had noticed the APC Mini stopped working a couple of times, last night.

I assumed a faulty cable, at the time, but could equally be the power ripple issue.

The pads were not lit at the time (obviously, since the APC isn’t supported yet).

toneburst, 2019-04-08:
Not sure if not being able to change pad brightness (as on LUnchpad) is going to prove to be a problem.

toneburst, 2019-05-08:
It was detected and worked fine when I last tried it. That would have been a few versions ago. I can try with the latest version this evening.

toneburst, 2019-05-08:
It was detected and worked fine as a generic controller last time I tested it.
That would have been several versions ago, but can test it with the latest version this evening.

toneburst, 2019-05-09:
I can confirm APC Mini works as a generic MIDI controller with the latest firmware (0.1.18).

The MIDI Monitor page is really useful, by the way!

toneburst, 2019-05-13:
The more I think about it, the more I think the lack of different pad LED brightness settings rules the APC Mini out as a Launchpad alternative.

I’d be really interested to hear your take on this, Simon.

westlicht, 2019-05-13:
Well, I didn't use multiple brightness values at first, so I think it could still work, even if it does show a little less information. But it's currently not high on my priority list anyway.

toneburst, 2019-05-13:
> Well, I didn't use multiple brightness values at first, so I think it could still work, even if it does show a little less information.

OK, that's encouraging.

> But it's currently not high on my priority list anyway.

That's fair enough. There are plenty of other exciting possibilities. I was just thinking I might not keep my APC, if you thought the single-brightness LEDs were a deal-breaker.

westlicht, 2019-07-31:
Moving this to the backlog.

closed 2019-07-31 by westlicht

toneburst, 2020-02-02:
I think I'm going to sell the APC Mini, as it's not been used for a while and I need to claw back some money for other projects.

If it's not going to be possible to freely assign the sliders in the Routing matrix (ie consume CCs from them as if it were a generic MIDI controller), while at the same time the pads are used Launchpad-style, then I'm not so interested in an APC implementation, anyway.