westlicht/performer #23: Auto-Increment To Next Step When Entering Notes Via MIDI

toneburst, 2019-03-31:
To aid fast note-entry, when using an attached MIDI keyboard to enter note, automatically increment the active step after entering a note.

I'd envisage starting entering notes in this way by holding a step button, and hitting a key on the keyboard (as it operates currently). After that, though, active step would automatically increment so the next key-press would set the note for the next step, etc.

One of the step buttons could be held in the normal way, to edit the note at that step, and auto-incrementing would continue from that step.

westlicht, 2019-03-31:
I do like the current ability to select a few notes and "transpose" them while the sequencer is running. I actually discovered that workflow during a live set :) and it was fun! I also like the fact that entering notes with the keyboard this way is very similar to how editing steps works on the sequencer as well. It's just more consistent. Entering steps as you describe could be a separate action.

toneburst, 2019-03-31:
Point taken. How about:

Pressing a step button drops into auto-increment mode.

Pressing currently active step button drops out of auto-step-select mode, returning the keyboard to transpose duties?

toneburst, 2019-03-31:
Also, auto-increment only works on Note Layer?

toneburst, 2019-03-31:
I think many people would find it quicker to enter notes this way, as you can do it without needing two hands.

westlicht, 2019-03-31:
One option may be to add another `RecordMode`. Currently there is `Overwrite` and `Overdub` which both do live recording. There could be another `Step Record` mode which would allow for entering a notes using a MIDI keyboard. So the workflow would be something along the lines of setting the `Step Record` mode. Then enter recording mode using `SHIFT` + `PLAY`. This would put the cursor on the first step of the sequence. Entering notes would write them to the sequence, wrapping around at the last step to the first step. Recording could be stopped by leaving record mode or restarted by leaving and entering record mode.

westlicht, 2019-03-31:
Did a quick proof of concept of the above idea and it works quite well. I think I might add it to the next release.

toneburst, 2019-03-31:
Cool, that should work :)
Good work!

closed 2019-04-01 by westlicht

toneburst, 2019-04-02:
Couple of enhancements that spring to mind:

- Wiggle pitch-bend to turn on Slide for that step (idea stolen from MI Yarns sequencer)
- Mod-wheel sets step length up to tied with full mod value

commit cdea54a3 (2019-04-01): added step record mode

fixes #23