westlicht/performer #30: Select Bus control (feat request)

james-everitt, 2019-04-09:
There's a growing number of eurorack modules that can store states and respond to preset recall via the 'select bus' spec. Malekko modules can, some WMD, some Make Noise, Expert Sleepers Distings and Mungo modules

This is really just a MIDI CC to indicate save or recall, followed by a midi program change message to indicate the slot. 

Here is a short video of a Teletype implementation im working on
https://www.youtube.com/watch?v=45sEcgE0N6Q

Here is a video from Make Noise
https://www.youtube.com/watch?v=W5mx6kuHgQU

These MIDI messages need to make there way onto the CV bus pin. Obviously that would mean a huge redesign, however in the mean time expert sleepers make a cheap DIY module that has MIDI input jack onto the bus.
---------------------------------------
I think the feature to send select bus control would be great to implement onto the Performer! Required MIDI messages are:

MIDI instruction on Project load:
CC 16 Val 64 (Load)
PRG change [Performer Project Number]

MIDI instruction on Project save:
CC 16 Val 127 (Save)
PRG change [Performer Project Number]

Those few lines enable save and recall of many modules states synced with the performer projects!

westlicht, 2019-04-10:
Good idea!  Is there a specification for this?

westlicht, 2019-04-10:
I found this: https://docs.google.com/document/d/1YhPvAI6oliwLYSHhDholdAUym-TfBDGlxST-zDCP-qw/edit#

james-everitt, 2019-04-10:
Yeah unfortunately Macro machines started the spec and documentation hasnt really been updated since.

The Make Noise Rene Mk2 manual has a schematic and Select Bus messages in its manual

[image]

I implemented it on the Teletype using a Teensy to take in I2C messages and output the required MIDI messages listed in that google docs

toneburst, 2019-04-16:
This is cool!

I've also been wondering if MIDI Program-Change and Bank-Select messages could be implemented for MIDI outputs. I will open a separate issue for that, I think.

westlicht, 2019-04-16:
Looking at the hardware specification, I don't think I will ever implement that. Outputting the MIDI messages might be worth looking into. But the spec is also a bit weird for my taste.

closed 2019-05-13 by westlicht

greengrooves, 2021-09-09:
Do you think it will ever be possible to send a prorgram change out to the CV bus like mentionerd above so that with the Expert Sleepers Select Bus Interface it would be possible to save and load Voltage Block and Disting Presets with the 16 slots of the Performer presets ? That would be such a great addiotion. Changed from a Varigate 8 to the Performer and the CV Bus communication is the only thing  that i  really miss.