westlicht/performer #31: Supervisor - ic or firmware

forestcaver, 2019-04-15:
On power up of my rack (doepfer psu3), performer fails to start until power cycled.  I am at less than 50% rated current for the psu. Lots of other digital modules. On other psus it starts first time. I suspect this is an in-rush current/timing event.

Suggestion: if you are updating the hardware - any chance of a supervisor ic on the rst pin? Or using the firmware supervisor? (But I know nothing about the stm32 built in supervisor other than it exists). Thanks for the consideration....

ob1techno, 2019-04-15:
Interesting. I had a similar issue using a Frequency Central Routemaster while testing one of my builds. I found that it failed to start about half the time when I powered the Routemaster with a 1A 12VAC wall wart, but worked 100% using a 1.5A 12VAC wall wart. Not sure if this is connected, but worth a mention maybe...

forestcaver, 2019-04-15:
Looking through the performer code - I am struggling to see if it is enabled - it’s probably my lack of understanding though :-)

westlicht, 2019-04-15:
I will look into it! Thanks for the feedback.

toneburst, 2019-04-16:
I have noticed the same issue, powering my Performer from an MI Module Tester.
It happens quite rarely but I have been it a few times.

jmsiener, 2019-04-17:
I have similar situation with my unit in the rack.  I have more than enough extra current between my two power supplies but it still fails to boot on first power up.  Nothing a power cycle doesn't fix but to avoid that would be pretty cool.

westlicht, 2019-05-06:
For now I implemented a watchdog in the firmware (v0.1.17). Please let me know if this fixes the the issues you are seeing. The next hardware version will also add some measures to hopefully improve this further.

closed 2019-05-06 by westlicht

forestcaver, 2019-05-17:
Still getting a fair number of episodes where it doesnt start up on power up.

forestcaver, 2019-08-19:
Hi Simon - I finally got my arse in gear and added a supervisor. It fixes all my problems on all my psus. It's trivial to add to the pads of SW8 (if it isnt soldered) and the +3v3 of R82. I also added a 100nf across RST and GND.