westlicht/performer #32: Step recording from cv input (implementation request)

Dunk303, 2019-04-15:
Would be possible use one cv input to enter note value in step recording? Would be super useful, as not everyone use midi keyboard\controller within modular.
Something like cv output from cv\gate keyboard into cv input 1 for note value, and trigger\gate out from keyboard into cv input 2 to determine when pass to the next step.

westlicht, 2019-04-15:
Could be a nice addition. I would probably implement it by a mode which uses CV1/2 inputs for cv/gate which then acts as if a MIDI controller is attached. This means it would also allow to "play" outputs when the sequencer is stopped and select note values in the steps page when editing the note layer. It would basically behave as if a MIDI controller is attached. Have to think about if this is easy to implement and has no negative side effects.

toneburst, 2019-04-16:
I guess because of the selected scale, you wouldn't necessarily have the same sort recorded as a VCO controlled by the same CV.

Would it also be necessary to calibrate the CV inputs in a similar way to the CV outputs?

Dunk303, 2019-04-16:
Yes of course recordings should reflect the selected voltage table. Or be unquantised :)
Also cv inputs should be sufficiently accurate, hope this is the case!

toneburst, 2019-04-16:
There's no dedicated ADC, so I guess the CV input will be limited by the ADC resolution of the MCU.

They're nominally 12-bit.

CV input range is -5 to 5V, which may not make them ideal for note CV purposes.

westlicht, 2019-04-16:
12-bit DAC should be plenty for tracking semitones. But it might require calibration, I haven't really looked at how bad the non-calibrated CV inputs are :) Adding calibration would be a good thing anyway.

westlicht, 2019-04-16:
Ok, did a quick proof of concept and the precision of the CV inputs is plenty for detecting semitones. I will probably implement this for the next release. I already have a feature in the pipeline that lets users configure the _Record Source_, e.g. from which MIDI port and channel it should record. Currently it always records from any source. This will allow to configure CV/Gate as a record source and then use this for anything a MIDI controller can be used.

Dunk303, 2019-04-17:
Terrific Simon! You are seriously kill the sequencer game with this one :) can't wait to test it out (take your time ofc)

westlicht, 2019-04-17:
This feature is almost done. I'm still a bit unsure about the following two options:
1. Add a _Enable CV/Gate Input_ option or similar to enable using CV1/2 inputs as CV/Gate, which then enabled generating virtual MIDI events from the CV/Gate source. This can then be used in all places as would a normal MIDI keyboard.
2. Add a _Record Source_ option as planned before. There are three versions for this, either just _MIDI_ / _CV/Gate_, _MIDI_ / _USB MIDI_ / _CV/Gate_ or the full set of _MIDI(omni,1-16)_ / _USB MIDI(omni,1-16)_ / _CV/Gate_. When _CV/Gate_ is selected, the translation would be enabled and only that input would be used for recording/monitoring.

I do tend towards option 1, as this is easier to use and typically you probably don't have multiple keyboards attached that you would want to differentiate. So simply enable the _CV/Gate Input_ option and be happy.

SloBloLabs, 2019-04-17:
In my use case a midi controller/keyboard would always be connected, either via midi or usb, but here and there I may want to capture a pattern from e.g. Metropolis. In these occasions a flexible input routing without repatching comes in handy. I would expect that cv1/2 would already be occupied by other input parameters like transposition or root note, so lets consider pair 3/4 for the feature, too.

Dunk303, 2019-04-18:
Well for me option 1 would be just fine, but if this could limit someone's flexibility option 2 could be better!

commit bb80b2e6 (2019-04-18): implemented cv/gate to midi conversion

Fixes #32.
closed 2019-04-18 by westlicht