westlicht/performer #35: MIDI Bank-Select and Program-Change For MIDI Output

toneburst, 2019-04-16:
How about implementing MIDI Bank-Select and Program-Change for MIDI outputs?

I guess the module would sent out the messages when first playing a Project.

To make it even more useful, bank and program-change messages could be set on a per-pattern basis, so patches could be changed on MIDI devices when changing patterns on the Performer.

westlicht, 2019-04-17:
Because performer is primarily a CV/Gate sequencer, not a MIDI sequencer, I don't think I will implement this immediately. It's just some more options that probably 99% of the users will never touch. On the other hand, people very serious about running MIDI setups will probably not use performer anyway.

toneburst, 2019-04-17:
That’s fair enough. I do think Performer provides a nice bridge between the standalone MIDI device and Eurorack modular worlds. It may not be the primary focus of the device, but it is a USP, I think.

Thanks for replying, anyway. I appreciate you taking the time.

westlicht, 2019-04-17:
Putting this to the backlog as I don't completely discard the idea yet :)

closed 2019-05-13 by westlicht

jackpf, 2022-08-07:
I've implemented some basic midi program change message sending here, if it helps: https://github.com/jackpf/performer/pull/10