westlicht/performer #36: Delayed Gates (feat. request

SloBloLabs, 2019-04-17:
Would be great if we could make sure the CV output value is available before gates are triggered. I encountered this not to be ensured when programming a certain pattern for BIA.
Current workaround: CV Update Mode = Always and programming the desired CV value in the note slot before (!).

westlicht, 2019-04-17:
It's something that I could add pretty easily. However, I will first investigate if the the CV outputs are actually behind the gate outputs, because that would be a general problem that would need to be fixed.

westlicht, 2019-04-18:
I've fixed the update order (no delay yet). Maybe that's all that's necessary. Will be part of the 0.1.15 release.

westlicht, 2019-04-21:
@SloBloLabs does the problem still exist with the 0.1.15 release?

SloBloLabs, 2019-04-21:
Simon, sorry for delayed feedback. Please bear with me. Can test the patch tomorrow. ;-)

westlicht, 2019-04-21:
No worries, let me know if it works, so I can potentially close this ticket.

SloBloLabs, 2019-04-22:
Hi Simon, thank you for putting effort in this. I reproduced the test case and it is not working yet on 1.15. Using my logic analyzer I could proove that CV is about 900us behind the gate.
Do you know wether the calls to the shift register and the dac are blocking ones? Are there known delays e.g. in the dac to come up with the voltage levels requested?

SloBloLabs, 2019-05-08:
Will test 1.17 by upcoming weekend.

westlicht, 2019-05-08:
@SloBloLabs I actually found the bug now. The problem was that the CV update was delayed by one system tick of 1ms. Thanks a lot for reporting this. This is actually a really bad bug to have on a sequencer :D will be fixed with the next 0.1.18 release.

closed 2019-05-08 by westlicht

SloBloLabs, 2019-05-11:
I had 1.18 on the board today. I confirm the gates came closer to the CV updates. I measure 15us time window inbetween, but gates are still hitting before the dac. Trying to fix it by introducing a delay within Engine::update():
    _cvOutput.update();
    for ( uint32_t i=0; i < 168 * 15; ++i) { // 168MHz, 15us
        asm ("nop");
    }
    _gateOutput.update();

It compiles fine, but the sequencer does not come up with this code. Same with
     _cvOutput.update();
    hal::Delay::delay_us(15);
    _gateOutput.update();

Do you have a clue why?

westlicht, 2019-05-11:
This is weird. I guess I have to measure again, but for me the gates were clearly coming after the CV changes. The delay should work (both your suggestions), does an unchanged version even work?

SloBloLabs, 2019-05-11:
Yes, sure. It’s just removing my additional lines, recompile and flash. Did this a couple of times today cause I couldn’t belive not being able to introduce such a small change. Need to add I measured with gate and cv connected to Basimilas. Shouldn’t make big difference, though.

SloBloLabs, 2019-05-13:
Simon, I confirm the issue is closed. The sequencer does exactly what I had in mind now. Thumbs up! Just ignore my previous comment. ;-)

westlicht, 2019-05-13:
Good to hear, I was a bit surprised at first when I saw your comment but also pretty confident it was all fine :)

commit 3144d500 (2019-05-08): fixed delayed update of DAC

fixes #36