westlicht/performer #38: MIDI Out

Shayshez, 2019-05-03:
1. There is few drum machines that respond only to midi channel #10.
Currently it's possible to work only with channes 1-8.
2. There is no sign at the midi monitor page.

Shayshez, 2019-05-03:
Another midi issue - gates are always short no matter gate length setting.

westlicht, 2019-05-04:
You can select the midi output channel by holding `SHIFT` and rotating the encoder to select the channel. I should probably change that behaviour and just have the encoder scroll through all the ports/channels. I think this gets people confused , even though it's stated in the manual. Does this solve your issue?

Shayshez, 2019-05-05:
Yes, i managed to change the midi channel, Thanks!
But there is issue with marged steps, if i set few steps to 100% length, performer sent midi note off after the first step, look at the video:
https://youtu.be/GyKOTGkxiKk

westlicht, 2019-05-05:
Are you using the latest version?

Shayshez, 2019-05-05:
Just built it in Thursday and installed v0.1.15, the latest version in the releases page.

westlicht, 2019-05-05:
Ok, will have a look then.

commit d53b5920 (2019-05-05): correctly handle tied notes on midi out

fixes #38
closed 2019-05-05 by westlicht