westlicht/performer #41: Root note error.

luisonly, 2019-05-08:
I´m having an issue when setting a scale (even user scales) and then changing the root note. For example if i choose "minor" scale and then choose D it will make make the note in my oscilator go to D# and then if i choose D# as my root note  the oscilator will output D#. It will quantize to wrong scale notes when changing the root note.

westlicht, 2019-05-13:
I cannot reproduce what you describe. Just tested this and it's all as expected. I will close this, but feel free to give me more detail on what your problem is.

closed 2019-05-13 by westlicht

luisonly, 2019-05-14:
Hi Simon, thanks for replying to my inquiry. Sadly the problem persists.
The simplest way I can rephrase my problem is this:

When I set my scale to:
Root Note C
Major scale
I get a C,D,E,F,G,A,B scale which is correct.

but When I set my scale to
Root Note A
Major scale
I get a E,F,G,A,B,C,D scale which is incorrect i should be getting
 A,B,C#,D,E,F#,G#

This happens with all the scales, even ones i make, it works perfectly fine
on root note C but messes up when switching the root note...

Is there something I did wrong? maybe in the calibration or something? i
double checked it and it seems correct.

Thanks for your time! and congratulations on such a great sequencer.

Luis Trasto.

westlicht, 2019-05-14:
Hi,

I just checked again with 0.1.8 and all is fine. Works with both the project wide scale/root note settings as well with sequence specific settings. I also tested MIDI output which also works fine. I'm really puzzled what is going on for you. Maybe you can setup a project with two tracks, both in different scale/root note settings and send the project file on the SD card. This way I could try to reproduce.

luisonly, 2019-05-15:
I havent tested with midi output, i will check it out and let you know.
Thanks for helping me out :)

westlicht, 2019-05-15:
@luisonly you can also try to make a video and send that to me. Just to make sure, you are running the latest firmware version right? If you do the test, please also start from a freshly initialized project, just to be sure. Should not matter but lets try to find out what's going on here.