westlicht/performer #42: key change in a song

hfrazer, 2019-05-08:
Would love to have octave&transpose to patterns to be applied at song level  so at a stage in the song the key can change.

westlicht, 2019-05-08:
Right now you can achieve this by creating a route to the octave/transpose parameter of **all** tracks you want to be affected. Then use one track (preferably the first one) and use it to automate that parameter. You will loose one track and setting it up is a bit fiddly but it can be done. You can then even sequence different transpositions by having multiple patterns on the first track and sequencing them in the song.

I have a work in progress implementation of a new track mode called "macro track". This track mode has the ability to sequence a special dynamic scale, e.g. by setting the scale and root note, inversions etc. This dynamic scale can then be picked up by other tracks just by selecting this dynamic scale. I would like to have multiple versions of this dynamic scale. E.g. one for bass, melodic and lead parts. This is still work in progress but it goes far beyond of what you can do with transposition. The downside again is that you have to dedicate one track for macro duties, leaving you with 7 tracks for outputting sequences. It would be nice to have this macro track as an additional 9th track, but this has far reaching consequences in terms of the UI. I'm not very happy with the current solution yet, but it's something I'm working on and I think has great potential, as I don't know any other sequencer with that kind of feature.

sneak-thief, 2019-05-11:
Great idea - transcending the UI's "8-track" layout will obviously be tricky.

What do you have in mind so far? 

1. My first thought is that it's not vital to have a "T9" LED as an activity indicator, so that's one less UI issue to worry about. 
 

2. What about using Page+S9 to enter/exit macro mode? It's a very fast & intuitive combo. S9=Special track 9 ;)

3. The performer page could have 9 boxes instead of 10 without any issue.

Using the OLED could be enough - and the Macro track could be turned on (bright) and off (dimmer) with the F5 key.

Macro On:

[image]

Macro On (alternative version):

[image]

Macro Off:
[image]

Macro Off (alternative version):
[image]

westlicht, 2019-05-13:
@sneak-thief thanks for the input. I don't think adding a 9th track is a good option. It completely breaks the design and handling of the UI. I still think that a new track mode is the better option. I also have some ideas for another new track mode which can output more than 1 CV signal. So this would allow you to use 2 CV outputs for one track, making room for the macro track. It's all work in progress in my head for now :)

sneak-thief, 2019-05-13:
Yes, definitely trust your instincts. Take your time and enjoy the process. I'm looking forward to using what you create and providing feedback.

toneburst, 2019-05-14:
I'm intrigued by this Macro Track idea, but I don't have a clear idea of what it might do.

will you start a new Issue thread to document your ideas on it, @westlicht?

toneburst, 2019-05-14:
I'm intrigued by this Macro Track idea, but I don't have a clear idea of what it might do.

Will you start a new Issue thread to document your ideas on it, @westlicht?

westlicht, 2019-05-15:
@toneburst yes, that sounds like a good idea. So people can actually get their ideas in before I start working on the final thing :)

toneburst, 2019-05-15:
That's cool :)

westlicht, 2019-05-17:
Moving this to #58

closed 2019-05-17 by westlicht