westlicht/performer #44: Dual Drum Mode To Use Both CV and Gate Outputs

toneburst, 2019-05-09:
How about a new Dual track mode, that could use both CV and Gate outputs to as drum triggers?

I would have two Gate layers, on for the Gate output, the other would use the CV output as gate/trigger.

toneburst, 2019-05-09:
Incidentally, I've discovered it's possible to use both Gate and CV outputs of a track as separate triggers for drum voice modules already.

If CV Update Mode is set to Always for the track, it's possible to use high and low note values to trigger drums from the CV output.

It's fiddly, though.

Might be easier if I made a custom scale with only one note in it.

westlicht, 2019-05-11:
I don't think this is an important feature to be honest. As you mention, it already is possible to do a similar thing by using the note layer as a second trigger layer.

closed 2019-05-11 by westlicht