westlicht/performer #45: Curve Track Randomisation

toneburst, 2019-05-10:
Some form of randomisation of Curve track steps.

The most obvious thing to randomise would be segment type.

Another possibility (maybe in addition to the above), would be some equivalent of the gate probability control for a Note track. In this case, how about a set-able probability to determine if the selected curve plays, or the value at the end of the previously-played segment is held for the duration of the current step?

westlicht, 2019-05-11:
You can already use generators to randomize curve tracks. But I need to put some work into generators soon.

toneburst, 2019-05-11:
That’s not quite what I had in mind (though cool).

I was thinking more of real-time randomisation, like for Gate/Note/Length on Note tracks.

westlicht, 2019-05-13:
Well, some probability functions might be nice on the curve track. Will have to think about it.

toneburst, 2019-05-13:
Sorry, yes, I meant probability, rather than randomising. My bad.
Random curve-track creation is also very cool, of course, but not what I had in mind.

I will change the issue title.

westlicht, 2019-07-22:
Curve variations will be part of the 0.1.27 release.

closed 2019-07-22 by westlicht

toneburst, 2019-07-25:
Very nice to see curve variation implemented!

It’s not quite what I had in mind- I was thinking something more along the lines of note randomisation in a Note track, ie the probability setting would set the likelihood of a random curve segment being played, rather than that a second, pre-selected type would play.

I haven’t had a chance to play with the current implementation, yet, so in practice it may work better than my idea.

westlicht, 2019-07-25:
In my opinion the note variation stuff is a bit too random for my taste. I might change that in the future. It works somewhat OK with scales, as you always end up on "good" notes, but in general I always felt there is not enough control. The curve variation layer on the other hand lets you create more specific variations. Have a go at it and let me know what you think.