westlicht/performer #47: MIDI Track Basic Arpeggiator

toneburst, 2019-05-10:
A basic tempo-synced arpeggiator option would be really nice for the MIDI>CV track mode.

westlicht, 2019-05-11:
This was in the back of my head for a while. It will probably get in at some point in the future.

toneburst, 2019-05-11:
Great! Looking forward to it.

toneburst, 2019-05-11:
I said “basic”, but obviously, things like play direction and real-time randomisation would be cool, too ;)

I will wait to see what you come up with, anyway.

westlicht, 2019-05-11:
> I said “basic”, but obviously, things like play direction and real-time randomisation would be cool, too ;)

hehe

closed 2019-05-17 by westlicht

toneburst, 2019-05-17:
Have you dropped this idea entirely, or is it incorporated into a branch you're working on?

westlicht, 2019-05-17:
It's about to be released.