westlicht/performer #49: Chords mode - enhancement

forestcaver, 2019-05-11:
Been thinking about how a chords mode could work. I cant decide if it is a sensible idea or not.

ob1techno, 2019-05-11:
Sensible or not, it’s certainly interesting!

westlicht, 2019-05-13:
@forestcaver can you provide more information, otherwise this issue does not make a lot of sense and I will probably close it.

forestcaver, 2019-05-13:
Basically, given the number of channels it would be great to be able to easily sequence chords rather than sequencing each note on multiple tracks to form a chord. I was thinking that there could be a library of chords (say just 4 notes) to act as a template for the other tracks that could be sequenced. The difficult part is trying to tie it in to the rest of the workflow which is what I am struggling with!

forestcaver, 2019-05-13:
(Cheekily I was hoping someone cleverer than me would have bright ideas for this!)

westlicht, 2019-05-13:
I do have some ideas for this. Its all in this _macro track_ mode I have mentioned before. I think it's a very cool feature that will set PER|FORMER apart quite a bit. I have an experimental branch with some ideas and it's a lot of fun to play with. But it's a pretty deep feature and I want to get it right before releasing.

forestcaver, 2019-05-14:
Excellent! I must have missed those comments on macro track! I agree anything like this will be tricky to get right and fit in seamlessly! Thanks!

toneburst, 2019-05-14:
@forestcaver https://github.com/westlicht/performer/search?q=macro&type=Issues

toneburst, 2019-05-14:
@forestcaver for reference:
https://github.com/westlicht/performer/search?q=macro&type=Issues

forestcaver, 2019-05-15:
@toneburst - thanks! I had actually seen those posts before but hadnt realised the implications of macro mode!!! Rereading them makes more sense now! Cheers...

westlicht, 2019-05-17:
Moving this to #58

closed 2019-05-17 by westlicht