westlicht/performer #52: finer percentage on Gate offset?

xhsi, 2019-05-14:
Hey first of all I want to say you doing great work with the performer. I am enjoying it a lot.
the gate offset is great, but it would be much more useful for me if I could set finer percentage like 3% 6% or so.

westlicht, 2019-05-14:
I would have added more precision if there were no memory constraints. See, the STM32 comes with 192kb of RAM. I have 8 tracks, each with 16 patterns of 64 steps in a project (plus clipboard space, screen buffer and a lot of other data). I currently have a limit of 6 bytes per step, which gives a total of 8*16*64*6=50kb just for the step sequence data. I'm thinking hard about every single bit I spend :) Also, I think the current resolution is great for sequences that have _adequate_ tempo, like for drums. If you set a very high divisor, then of course the offset steps become larger as well. But it's just a tradeoff I had to make, sorry.

closed 2019-05-14 by westlicht

napoleonthegreat94, 2020-05-31:
Thank you for this great sequencer! Personally i would give up the whole rerigger layer just to be able to have finer percentage on gate offset. No complains and i understand that it would mess the UI but i think from a musical perspective it would make a HUGE difference! Being able to randomize or program the gate offset with precision would be something else! I find that getting away from the time grid in a controlled manner just like musicians do playing their instruments would be a big leap forward.  
Thank you for this awesome project

westlicht, 2020-05-31:
@napoleonthegreat94 how much additional precision would you need?

napoleonthegreat94, 2020-05-31:
@westlicht i guess there are two different options, a step of 6.25 (16 steps per note) or a step of 8.35 (12 steps per note). These steps series would both go through 25,50,75 and 100 percent. 

i think that a step of 6.25% would be best!

Thanks for the fast reply

napoleonthegreat94, 2020-07-07:
so i saw another way to go about it, from colin benders on utube and thought to mention the idea here. you need only a lfo (or oscillator) an attenuator and one cv channel from the performer. the lfo is clock and feeds the clock in of the performer. the one channel from the performer is sequencing the clock-lfo in order to change the pulse length and effectively the feel and groove. attenuating the sequence of the clock can give great resolution on the timing . with reset out you can sync a second sequencer which doesnt have to share the same 'wonky' clock.