westlicht/performer #53: affecting divisor on routing page

xhsi, 2019-05-14:
would be nice if One could also affect Divisor on the routing page

westlicht, 2019-05-14:
I haven't added this because it is a bit tricky. What divisors would be addressable? Just multiples/divisions of 2 or every possible divisor? It might be fun as an experimental feature I guess.

toneburst, 2019-05-14:
I've come across similar issues with other products implementing MIDI control.

Sometimes, being able to set a simple range isn't enough, and what you really want is the ability to define exactly which values can be selected.

It's a challenge to come up with a UI for defining these values, though, especially within the constraints of the hardware.

xhsi, 2019-05-15:
I imagine it like on the routing for last and first step.
you only choose the lowest and highest divisor and then cycle through the in between. 
I think its important to adress every possible divisor.
for exact selection, I guess one has to use the right cv source 😃

westlicht, 2019-05-15:
I have just implemented this as a proof of concept. It revealed an issue with the clock division. The current implementation of the clock division uses modulo instead of counters, which leads to "jittering" when changing the divisor smoothly. I have to think about this for a bit and decide if it's worth changing. But in the current state, I think people would not be happy with it. This is not really a problem when divisors are preset and don't change all the time.

xhsi, 2019-05-16:
I understand that .
anyway thanks for making that awesome module!

westlicht, 2019-05-17:
To be released with version 0.1.19.

closed 2019-05-17 by westlicht