westlicht/performer #55: Note Mode Display Note Names

toneburst, 2019-05-15:
In the Note layer of a Note track, it would be great to have the option to display note-names on the Performer display instead of scale indices.

Obviously, this would only work with the chromatic scale, and other scales that use "standard" tunings, but in these cases, I think many would find it helpful to be able to see the name of the note, as well as the octave.

westlicht, 2019-05-16:
What if the sequence is transposed or using different root note?

toneburst, 2019-05-16:
..then the notes shown won't be the notes that play. Equally, the indices shown currently won't be correct, either.

I still think it will help a lot when entering notes initially to be able to see the note and octave.

toneburst, 2019-05-16:
On a related theme, it would be cool if the current root-note and transpose value were displayed on the Steps screen, I think (in the Header bar, would be the most logical place).

sneak-thief, 2019-05-17:
I'm also in favour of the option to show note names instead of numbers. 

I understand the pragmatism and flexibility of defining the Performer as relying on a "concept of voltage tables". 

However, a good portion of the Performer still relies on chromatic Western notation to get the job done, eg. semitones, midi notes & most of the scales. As a musician, numerical notation is often fine, but there are often times when note names are very welcome, especially during deliberate composition.

Should the note layer account for transposition and root notes? I guess that depends on how hard the task is and how much RAM it would require ;)

westlicht, 2019-05-17:
Next release will have note names again. It takes root note and scales into account. Transposition and octave switching is not taken into account though, as this might lead to pretty weird results when automating those parameters.

As for the suggestion to show the root note and transpose value in the header, this won't happen. I feel it breaks the UI too much. I would have to change things more drastically (e.g. replace the page and context text with a smaller form in order to get enough real estate for additional information).

closed 2019-05-17 by westlicht

toneburst, 2019-05-20:
This is great, thank you, Simon!

commit d1feb406 (2019-05-17): show note names when using chromatic scales

fixes #55