westlicht/performer #56: Alternative Note Off method not supported

sneak-thief, 2019-05-16:
PROBLEM

The alternate midi Note Off command (Note + Velocity 0) isn't recognized by the performer and a previously played note stays on. 

"The MIDI Specification says:

"A receiver must be capable of recognizing either method of turning off a note, and should treat them identically." "

https://www.midi.org/forum/228-writing-midi-software-send-note-off,-or-zero-velocity-note-on

I tested this with my Yamaha Montage. 
 

EXPECTED BEHAVIOUR

Sending midi "Note + Velocity >0" followed by "Note + Velocity 0" should turn the note/gate off.

westlicht, 2019-05-16:
Ok, didnt know there are devices not sending note off events. Should be easy to fix.

closed 2019-05-17 by westlicht

sneak-thief, 2019-05-21:
I confirm this fix works. Thank you!

commit 36837e45 (2019-05-17): handle fake note off events

fixes #56