westlicht/performer #58: Harmony track mode

westlicht, 2019-05-17:


toneburst, 2019-05-17:
Is this the "Macro" mode you mentioned before, or something else?

toneburst, 2019-05-20:
So new Track Mode would allow you do define chords on a per-step basis? Would you be able to control the number of available voice channels (ie CV/Gate pairs) globally?

westlicht, 2019-05-20:
> So new Track Mode would allow you to define chords on a per-step basis? Would you be able to control the number of available voice channels (ie CV/Gate pairs) globally?
> 
> Would activating this track mode on a track effectively deactivate 3/4 etc. preceding tracks, so they become simply outputs for the chords generated by the Chord track?

No, the chord track would not output any cv/gate signals at all. At least that was my initial idea. It would only control dynamic scales which can be used in other sequences. Think of it in terms of the currently available scales. You can set a sequence to use a certain scale and root note, which maps the notes to the notes of the scale. With the new track mode, you could basically control which notes are in the scale. You would do this by setting chords rather than scales. So the dynamic scales are not necessarily the same as a typical scale, but rather a selection of notes in the chord.

toneburst, 2019-05-21:
So for a particular channel, using one of these dynamic scales, would the scale only have one note in it, so it would play one note in the chord selected in the Chord track?

Would that mean that you could only turn the gate on/off and set the octave, for that track's notes?

westlicht, 2019-05-21:
> So for a particular channel, using one of these dynamic scales, would the scale only have one note in it, so it would play one note in the chord selected in the Chord track?
> 
> Would that mean that you could only turn the gate on/off and set the octave, for that track's notes?

No, imagine you create a sequence with Cmin, Fmin, Gmin7 (or whatever). For Cmin it the scale would contain the notes C,Eb,G. Depending on attributes it would also add octaves of it, or use inverted form, adding only base octaves etc. Fmin would contain F, Ab, C and Gmin7 would contain G, Bb and D and the seventh. Now, a sequence playing notes 0, 1, 2 in a loop would simply play the triads. A sequence with just note 0 would start on C, then switch to F and finally to G.

I have not fully worked what options there would be to affect the scales, but this is the basic idea. Create simple patterns on a note sequence and then have them play on "scales" that basically allow them to play arpeggios other melodies on the notes of a chord. This way, you can program longer harmonies using a slow divisor on the chord track and have shorter repeating sequences that move to different chords.

Does this make more sense?

toneburst, 2019-05-21:
Still a bit confused. It does sound cool, though, so I'm looking forward to giving in a try, when it makes it in to a release.

Funnily enough, after having missed out on basic music theory before I started making music (and more-or-less managing by ear), I've just been reading up more about chord types, so this development comes at a good time for me ;)

toneburst, 2019-05-22:
I think I was over-complicating things in my mind. So the Chord track literally just creates a custom scale based on the selected chord-type and root-note for a given step, and makes that scale available to other Note mode tracks.

I had it in my head that the Chord mode track was somehow controlling several Note mode tracks directly.

Will the dynamic scale from the Chord track to selected on the Sequence page, like any of the static scales? I guess there would need to be a reference to the index of the Chord mode track, as you'd be able to have multiple Chord mode tracks, potentially. Or would you?

Things might get confusing re. octaves, with chord shapes that span multiple octaves. Or will the dynamic scale override the octave setting of Note mode track step, in order to ensure that a note in a lower octave doesn't end up b being transposed up an octave, or vice-versa?

toneburst, 2019-05-22:
It might also be cool to have only the root-note selected in the Chord track applied to a given Note track, rather than root note and dynamic chord scale.

That way, it would be possible to create a bassline that was automatically transposed up and down to follow the chords, but was also able to use notes that aren't in the custom scale for the current Chord mode chord scale.

toneburst, 2019-05-22:
I'm intrigued to know what the proposed new multi-output track mode is, too :)

UncleRoel, 2019-05-23:
I love this idea. I'm a jazz harmony (amongst other subjects) teacher by day, if you want to discuss chord types to implement, you're more than welcome. 

I've made an application for O_C hemisphere that implements a simple arpeggiator using different "chords types". In this app I also added plain octaves and fifth+octave as "chords". This allows for typical arpeggio patterns to be created. (Code in my github, long time since I updated it.)

On parameters: the ability to inverse chords would be great. (Example root C, min7 type, 2nd inversion.)

jmsiener, 2019-05-29:
I’m digging what @uncleroel is saying here. I understand your perspective of arpeggios in an array but chords stuck in a natural minor scale doesn’t have much in the way of “functional harmony.” It’d be great for making a deep house track where every chord is a parallel m9 but other tunes in minor key are probably going to have a major 5/V/dominant chord.

Really love this idea, would be great for routing to midi to something like threshold/uEdges. If it was even simpler like sending up to 3 intervals (root+3) in a step then it would be great for a midi-drum trig module like dodeca/dubldeca. Really glad you’re thinking of a way to implement poly to the sequencer, really want this.

UncleRoel, 2019-05-29:
Reading the original post again and pondering; I think @westlicht is on the right track here. If the scales are variable per step, this would allow exactly those things.

With variable scales; the parameters per step could be something like:
- scale
- scale degree (optional, the root of the chord within the scale. This would allow to very easily make traditional progressions: you select the major scale, then set a I VI IV V sequence)
- chord structure (triad, seventh chord, add9, sus4, sus2, ... but also adding just octaves + fifth)
- inversion

Allowing other tracks to follow the chord track with a selection of following just root, bass note (in case of inversion), chord or full scale would make it a tremendous tool. Add to that the possibility of having more than 1 chord track in the system and the possibilities would be even beter. (Allowing for chords with upper structures (stacked chords), it's a fairly common practice in jazz. Or different sets of scales to be used over 1 set of chords.)

toneburst, 2019-05-29:
@westlicht will these Dynamic Scales always contain the same number of notes?

I can see things getting confusing, if they don't.

For example, if a given step on a track using a dynamic scale defines a note with an index of, say, 5, and the per-step dynamic scale defines a scale with only 3 notes, you're not necessarily going to get the note you expect.

westlicht, 2019-05-29:
@jmsiener this track mode is not intended to output polyphonic signals. The main idea is to allow sequencing a progression by controlling a set of dynamic scales which can then be used by other tracks to "play within the progression". If you'd want to output chords (e.g. using multiple CVs) you'd have to setup multiple tracks outputting a single note each. But by using the dynamic scales you could have one track output the 1st note in the chord, the second outputting the 2nd etc. But the new track mode itself will not output anything but just control these dynamic scales.

@UncleRoel I still haven't decided how to define the chords yet. But my main idea was to specify a chord root note and the chord quality (e.g. a whole list of chords). Maybe it would be easier to split the chord quality into "properties" instead of just a list to scroll through. Also I'm note sure yet how to link the scale to the actual chords, because only using the notes in the scale for building the chords is limiting. But maybe there could just be a scale that is set independent of the chords. Also, regarding the question about stacked chords. Maybe having just two chords per step would solve that (of course adding a second set of dynamic scales). But maybe this is also overkill in the beginning. I'm also wondering how this feature could be made more user friendly. E.g. choosing chords that work together as a default and allow deviating from it if needed.

Well, there is a lot to still think about. I have a prototype and it tells me there is quite some potential in this idea, but I still have a lot of work to make it easy to understand I guess :)

UncleRoel, 2019-05-30:
The most user friendly way to generate chords is to have them derived from a scale rather than ordered by chord quality (m7 vs Maj7 etc....). Doing this in any major or minor scale will give you usable progressions (although not necessarily in that key). 
When doing this you can use generic chord structures to derive them. For example, root-third-fifth for a triad. Depending on the degree within the scale the third will be major or minor, but always within the scale. (Scale degree = the note number in the scale on which the chord is built, mostly notated in Roman number. For example, 2nd degree in the scale of C major is: Dm)

If you can choose the scale root and scale for each step together with this, you can do most of the tonal and modal music in a fairly easy way. 

I would suggest adding harmonic and melodic minor scales to be complete. For ease of use adding the modal scales might be helpful (although not at all necessary as changing the root to another major scale is the same). 

If you choose to select chords by chord quality, it requires basic theory knowledge of the user to get a diatonic progression and more settings will have to be done to dial it in. It's also in the hands of the user to find the scale that fits the chord(s) at that point.

UncleRoel, 2019-05-30:
On the scales: building chords only from the scale is limiting indeed, but I would suggest having the scale and root being variable per step as well. 

On stacking chords: if you program it in a way that allows for more than 1 chord track you can easily stack chords or use one track to derive chords from and another one to derive melody from. 
Having more than 1 chord in a track would make it confusing for most users.

Feel free to contact me to discuss chords etc further.

UncleRoel, 2019-07-02:
I was browsing around and discovered that the conductive labs NDLR uses the scale degree system I was referring to. If that is implemented with a selection of scale and scale root at each step, it becomes a powerful tool. Easy of you keep scale and scale root the same, but flexible of you don't....

dComposer, 2019-08-13:
The immediate impression I got when I first read about this chord track was something like what's in the below screenshot. You'd be able to lay out the chords you want to use and how many beats each chord is to play for and then it would be cool to couple this track with either your arpeggiator or your newly built midi filtering capabilities (to tease out a bass line) to perform on top of:

[image]

shinito, 2019-08-21:
I’m very interested on this! I’ve been trying to set a proper workflow working with chord/scale sequencing (with Instruo QPQ). The main not solved issue I faced is the fact that chord sequencing is not giving the whole information needed to build melodies, and scale is not giving specific harmony information to properly set a chord progression.
Related to what was already commented, I got the idea of a system in which you can define the tonality and harmony in each step, so the whole scale mask is available to build melodies and chord progression is as well to build a harmony background.
This could be achieved maybe working with parameters per step like these:
Scale (defining a tonality, you’d change this for modulation or modal interchanges)
Chord Grade (within that scale, so chord quality gets automatically defined)
Tension Mask (defining the notes used to buid the chord)
The Tension mask options would be something like root, root+5th, triad, 1357, 1356, 12357 (13579), or a custom user mask.
So, this way, you could set different ratio between chord and passing notes adding much more musical sense, setting certain notes (those in downbeat steps or whatever else criteria) with higher probability for notes within the chord than the rest of the scale depending on what kind of track you’re building (lead melodies, basslines or chord background).

froschmett, 2019-09-29:
Hello,
could this also be used as an arp for CV tracks? Like defining the root note and type of chord?

dcbb, 2019-10-24:
Just stumbled into this – a harmony mode would totally make switch to this module.

Regarding implementation, I recommend looking very closely at the ACL Sinfonion (nicely explained by its creator in a series of Youtube vides), which has a couple of great ideas for implementing deep multi-track harmony. Sinfonion is basically a multi-channel quantiser with a chord progression sequencer, and a beautiful implementation of the ideas @shinito points to.

Especially, I think it's important to avoid a distinction between chords and scales, in the sense that – for example – the notes of the C major scale are just the same thing as the notes of a wild C maj7 +9 +11 +13 chord.

I think scale + degree + tension mask, as pointed out by @shinito, is the exactly the way to go!

walterschulze, 2020-01-14:
Hey. Is it possible to have Scale as an option to assign to cv input? 
Almost everything else is there already :) 
And I feel like this might also be a step towards the issue being discussed here.

walterschulze, 2020-04-30:
Would it be difficult to 
Store Polyphonic Midi Input across multiple channels?

i would like to designate individual amounts of tracks, for example Track 1-4 
to type CHORD. in Routing. So like Midi, but different - 
Meaning when i play midi keyboard,
whichever track 1-4 is selected, the input gets recorded polyphonic. 
equally distributed across track 1-4.
when i select track 1. i can see the notes that ended there.
when i select track 2 i can see the notes that ended there etc.
that way you don't need a "chord" system to be stored per step.
and cv output still output monophonic data but up to 8 outputs.
yes? please? :)

westlicht, 2020-05-03:
@walterschulze your suggestion is not really in line of what I had in mind with the harmony track. So I don't think I will work towards the feature you requested, sorry.

I actually started thinking about the harmony system again the last few days and have some ideas of where to start. It will actually not be a new track mode but a separate page similar to the song page. On that page you will find a lot of options with one goal only, which is to control a set of _dynamic scales_. These scales, similar to the ones currently available, are just voltage tables which the sequences can index into. The difference is that those scales are dynamic and change with the settings on the harmony page. I will draw quite some inspiration from the excellent ACL Sinfonium module which actually shares quite some common ground of what I had in mind initially over a year ago. The main idea is to derive the dynamic scales from a root note + chord/degree and then select what notes (root, 3rd, 5th, 7th etc.) go into a dynamic scale. There will also be options for inversions, A/B parts and other things. Lots of things think about, but I'm getting excited about the possibilities (as I was before, but just was a bit distant for a while).

shinito, 2020-05-03:
Oh, great to hear you're back in thinking about it!
So, you mean root+scale to define a key, and degree to define chord, right? And then, you could chose a sequence to index into either the whole key scale or the chord mask? Is that your idea?

westlicht, 2020-05-03:
@shinito yes that's part of the concept. But I think there are going to be two ways to define chords. One is as you suggest using a key (root/scale) and then a degree to define the chord. The other may be more explicit by choosing a root + chord quality directly. This will allow more flexibility in programming more elaborate chord progressions.

shinito, 2020-05-04:
@westlicht Well, that'd be a more direct way to define a chord. Both systems would lead to the same results actually. I mean, if what you mean by "elaborated chord progressions" is progressions including non diatonic chords, these would be achieved dynamically defining new root/scale, as this is what actually happens when you use a non diatonic chord: you do key modulations or modal interchanges, and every chord out of the first key can be seen as a degree in a new root/scale definition. This can be tricky for some people, depending on how they see harmonic changes, as you need to think what key a new chord belongs to, but it's more musical to me.
Having that second method available would be great, but then I guess a root/scale and a chord out of that root/scale could be available simultaneously, right? In a way that a track could be indexed into a scale while another one is giving a different key's chord.
Interesting. Really excited about this ideas!

toneburst, 2020-05-04:
Good to see you’re working on this again, @westlicht! Excited to see what you come up with.

walterschulze, 2020-05-05:
glad to see you're working on it again too :) 
I've had a look at the implementation in the sinfonion module... wow.
now i'm curious to see your vision put to form.

i've had great success in building complex polyphonic sequences and with the 4 cv inputs, changing octave, transpose and now even switching between custom scales, as well as pattern rotation over CV to achieve some already mindbending results. 
don't mean to be the bad guy, i just reckon it might be even crazier with polyphonic note input enabled somehow to lay some chords down quickly by hand (i have the novation pad on the usb midi as well as a midi keyboard on midi input always) - just a fantasy.

one thing i haven't found without sacrificing an extra track: 
the song mode window doesn't have a "transpose" column to transpose patterns after chaining them - would this be something to be included in the new window? just food for thought.

nemoy, 2020-07-27:
Hi @westlicht;
I Love your chord Progression Idea! And here's a little Idea of mine too, regarding the fact, that you'd have to sacrifice an entire Track for that: 

What if you could have **Midi In** controlling all the chords? Not using up any track data at all, but use whatever is currently present at the midi in directly to change scales and rootnotes. In the process never writing anything onto any track. That way you don't sacrifice a Track, but still have a very powerful tool for performing chord progressions.
(you will have to play them on a midi keyboard, or chordbot or something similar though, but I'd love that! : )

Please :)))