westlicht/performer #60: LFO

sinneb, 2019-05-21:
Loving the westlicht but I'm missing some quick LFO functionality.

A way to implement this would be to introduce "LFO" as a track mode. Then, under "steps" there's a menu simular to the MIDI/CV mode menu with the following options "waveshape", free running / tempo synced" and "sync multiplier / divider". The CV out for the selected track than outputs the LFO.

Future updates can be the inclusion of single cycle waveforms, linking LFO tracks to combine waveshapes, drawing of LFO's like the functionality under curve.

or...

this could be a generator under the Curve mode. The LFO generator would than co-exist with the Euclidean and the random.

toneburst, 2019-05-21:
You can already create a tempo-synced LFO using segments in a Curve track.

That said, there is a limit on how slow a segment can play back, relative to the project tempo.

One of the proposed enhancements is the ability to allow curve segments to span multiple steps, though, so with that in place, creating a wide variety of repeating LFO shapes would be easy.

westlicht, 2019-05-21:
I also think you can achieve most of that using the curve track. Just make a 1 or 2 step sequence and play with the divisor. What I might add in a future update is even larger divisors to have even slower sequences :) the "curve tie" feature is also in development, although I haven't found a sweet spot just yet.

toneburst, 2019-05-21:
> What I might add in a future update is even larger divisors to have even slower sequences

Yes, great!

sinneb, 2019-05-21:
I just experimented with the proposed setup in a curve track and that's working perfectly. Slower sequences are even better so looking forward towards the larger divisors.

closed 2019-05-21 by sinneb

toneburst, 2019-05-21:
@sinneb in the (current) absence of larger divisors and extended curve segments, you can set a low tempo (say, half the intended tempo), then set smaller divisors (ie 0.5x the normal size you'd use) for your non-LFO tracks. That way, you can get even slower LFOs.