westlicht/performer #62: Novation launch control implementation

thopa, 2019-05-22:
Would be great to be able to sequence notes on/off and pitch with the novation launch control pots and pads.

Also if there was a customizable way of having specific control changes midi messages control sequencer parameters. 

Thanks for a great project!

toneburst, 2019-05-22:
Are you talking about the Launch Control, or the XL version?

thopa, 2019-05-22:
Hey A!
I was referring  to the launch control. I also  have access to the xl and i think the midi implementation should be quite similar right?
Cheers

westlicht, 2019-05-22:
I'm not too fond of that controller layout. In contrast to the launchpad, it will be hard to make something that is sort of able to control most aspects of the sequencer. Also, I don't think using the pots for entering note values is a great experience as they are not encoders, but just rotary pots. You don't get haptic feedback on when you change from one note to the next. I think using the encoder on the module is way better for that (also it lets you edit multiple notes at the same time). Or connect a MIDI keyboard and enter notes in step record mode, live record mode or just by selecting steps and pressing a key on the keyboard. Again, I don't think those controllers add much for editing. They are great controllers for the routing system though!

toneburst, 2019-05-22:
Hi Thopa!

I'm with Simon on this one, I'm afraid. Both are great controllers, but they would imply a very different way of working, if they were to be used for sequence editing on the Performer, and I'm not sure the extra effort working out a system to use them would be worth it.

thopa, 2019-05-22:
Thanks for the replies and for the feedback, much appreciated.
I understand that the suggested controller might not have the best layout for controlling the performer, but how about being able to assign midi messages to performer functions? This way one could make the layout that most fits his| her needs?
Thinking of Lemur, or other customizable  midi controllers software / hardware.
Thanks!

westlicht, 2019-05-22:
> I understand that the suggested controller might not have the best layout for controlling the performer, but how about being able to assign midi messages to performer functions? This way one could make the layout that most fits his| her needs?

Well, you can use your MIDI controller for controlling routable parameters, which is really cool for performing. However, editing steps is far beyond that. How would you map those? There are way too many parameters for that. I added support for the launchpad controller because its layout is well suited for editing sequences, launching patterns etc. I might add support for other grid controllers, but generic MIDI controllers will _only_ be able to control routable parameters.

thopa, 2019-05-22:
Yes thats true Simon. Implementation of midi note, length , velocity, and mapping these controls  to an specific controller could actually put limits to what performer its really good at, which is hands on control of the sequencer. The implementation of note data would also be limiting too to the controller.
Looking forward to the new features. Cheers

closed 2019-05-22 by westlicht