westlicht/performer #64: Feature request: set parameters using midi cc

UncleRoel, 2019-05-23:
A great enhancement would be the ability to set the parameters of the selected track/page (in the steps page) using midi CC, allowing to set them with any midi faderbox. (16n being ideal for this, 16 faders and open source and available for DIY)

This would, for example, allow to set a note sequence in an analog sequencer style. The sliders select the note index out of the selected scale. (For this, a setting to select the range of the note parameter would be preferable. Other parameters should be fine.)
But it would also allow easy editing of the curve shapes, length, .... All parameters would be editable without needing 2 hands!

A sequencer setting that selects which input channel to take the midi CC from (usb or midi in) and base channel for the 16 CC parameters would complete it.

UncleRoel, 2019-05-23:
Elaborating further: 
Using note on and off messages to toggle the gates in step mode would allow for using a midicontroller that has faders and buttons to be used. I can imagine someone finding time and courage to design a module with 16 faders and switches to be a companion for the performer. Especially since the 16n already provides a good starting point.

toneburst, 2019-05-29:
Unless you have "flying faders" or endless encoders and some form of LED-ring orscreen to show the current value, the actual value of a given parameter is not going to correspond to the position of the knob/slider, most of the time.

That't why the Launchpad works so well for editing step data, and other controllers with rotary potentiometers or sliders are a much more awkward proposition.

Providing the necessary MIDI control scaffolding might still be useful, but I'm not sure the perfect hardware to make use of those functions exists, currently.

@westlicht might have a different perspective, though.

UncleRoel, 2019-05-29:
Even without the visual feedback. You have the feedback on the display, it would just allow for an easier method of entering values. Especially when setting up a sequence at first. It's a very natural way of entering values...

Of course, you can go into a complete midi implementation where you have CC linked to each specific parameter with midi channel corresponding to the track nr. That would allow using a BCR2000 or BCF2000 to be setup. But that might be overkill. Just having control over the selected steps in the selected track would be neat.

toneburst, 2019-05-29:
@UncleRoel I’d forgotten about the BCR2000. It’s aesthetically unappealing, and no longer in production, but might still be pretty cool.

So would the MIDI implementation include notes or CCs for selecting channel and pattern page, notes for 16 gates, and CCs for the step layers of selected page?

UncleRoel, 2019-05-29:
That's up to @westlicht to decide, I guess. 

In any case, a setting that allows for a basic control of the current parameter (maybe in addition to full midi implementation) would get my preference. I have both BCR and BCF2000, and besides being ugly, they are also quite big to have on the desk.

toneburst, 2019-05-29:
I wouldn't be surprised if @westlicht has some ideas for a dedicated control-surface for the Performer.

Having said that, the Launchpad does work really well, and I'm sure he's going to continue to refine Launchpad integration in the future.

I'm really hoping Novation will eventually release a full-RGB version of the LP Mini, because I'm a sucker for multi-coloured illuminated pads ;)

Think I'm drifting OT, though...

westlicht, 2019-05-29:
I have mixed feelings about this. My design from the start was to support external control surfaces, but I think making this a generic thing will quickly become messy. Sure, you could define a fixed array of CCs to control aspects of the sequencer but it would not be as well integrated as with the current Launchpad implementation. That being said, the control surface implementation allows to add other controllers in the future, but I think it makes more sense to support few devices very well instead of having support for general controllers.

westlicht, 2019-05-29:
And no, I don't have plans on creating my own MIDI controller tailored to PER|FORMER. I think the current interface on the module itself is pretty nice.

toneburst, 2019-05-29:
> I think the current interface on the module itself is pretty nice.

Agreed :)

UncleRoel, 2019-05-29:
How do you feel about controlling the selected parameter on the step page using the 16n faderbox? This would allow fast editing without needing 2 hands (hold the step button and turn the encoder).

UncleRoel, 2019-05-29:
(That was my original intention here, it drifted of from there towards a complete midi implementation.)

westlicht, 2019-05-29:
I don't think absolute controllers (faders, rotary pots) are nice for that, because you almost always have a discrepancy from what the steps are set to and what the faders/pots are set to. Then you either jump to a completely different value if you touch a controller or you have to _grab_ the value by moving the controller through the currently set value. It's just not the same as having bidirectional communication. You can only get this with endless encoders. In addition, I prefer a control surface to be operable without having to go back and forth to the module. This is what the Launchpad implementation gives you, it allows to control many aspects of the sequencer without touching the module.

toneburst, 2019-05-29:
As I mentioned above, some kind of custom controller with endless encoders and either LED rings or a screen to show current values when switching pages, tracks etc. would be cool, but the perfect hardware doesn't currently exist.

westlicht, 2019-05-29:
> some kind of custom controller with endless encoders and either LED rings or a screen to show current values when switching pages, tracks etc

Thats basically what the module is :) sure, i made the decision to put only one encoder on it, but this was a very conscious decision. There are other sequencers like Five12 Vector which might fit the bill for this kind of workflow much better.

UncleRoel, 2019-05-29:
@toneburst This kind of controller does exist: 16 encoders with led feedback. Fairly expensive though. The led ring feedback might not be very suitable for all parameters either.
https://store.djtechtools.com/collections/midi-fighters-midi-fighter-accessories/products/midi-fighter-twister

westlicht, 2019-07-02:
I just found out that the 16n faderbox has midi over usb. So it would be a good candidate for a new controller type. If someone wants to provide me with one of those boxes I will implement it. Otherwise someone would have to provide me with the usb vid/pid as well as the cc channel numbers of the faders. That should be enough for an initial implementatiob, but having the box would allow me to tweak it towards something better :)

UncleRoel, 2019-07-02:
Great, I haven't build mine yet. I can get a move on so I can test things. 
The 16n is open source and runs on a teensy. I'm sure finding out what the midi cc numbers are will be easy. 
I'll look into it tomorrow.

westlicht, 2019-07-02:
Yep, already found out. The problem is that teensy uses a fixed pid/vid (haven't found it yet) but this would mean that any teensy based midi usb thingy would be detected as the same controller. There is a device string that can be set, but performer does not currently read that out.

UncleRoel, 2019-07-02:
One possibility could be providing a hex for the 16n that does change the pid/vid? Or is that further down in the bootloader?

And otherwise, what further implementation were you thinking of? Because opening it up for all controllers sending cc is another option. (And that brings us back to the previous discussion.)

westlicht, 2019-07-02:
I don't like the idea for generic controller support. It should be plug and play.

westlicht, 2019-07-31:
Moving this to the backlog.

closed 2019-07-31 by westlicht