westlicht/performer #65: Feature request: Track mirror/morph

sneak-thief, 2019-05-28:
Track mirror/morph would be Track option where you could make one track (destination) morph into another track (source) from 0-100%. 

At 100%, the destination would have the same notes & gates as the source (aka mirror).

You would be able to select the source track and as you increase the percentage, the notes of the destination track would shift towards the source track. 

The gate distribution/density and length of the destination track would also shift in a probabilistic method towards the source track.

Transposition, scale and root notes of the destination track would override the source track.

Bonus points if the percentage value could be routable.

toneburst, 2019-05-29:
+1 for this, but a lot of thought would need to be put into the implementation, to produce musically useful results, I suspect.

sneak-thief, 2019-05-29:
From my perspective, the "musicality" of the function should be agnostic considering the vast musical sphere. Guided "happy accidents" is what makes this function desirable. 

It really doesn't have to be a complex implementation for it to be incredibly useful.

That said, I suppose one could have the granular option to morph 1) notes, 2) gates and 3) gate lengths.

toneburst, 2019-05-29:
> That said, I suppose one could have the granular option to morph 1) notes, 2) gates and 3) gate lengths.

I think that would be a good idea.

westlicht, 2019-05-29:
I think this is an interesting idea. But I would definitely implement it as morphing from one pattern to the next on the same track. Otherwise its becoming a mess.

One possible implementation would be to extend the _fill_ feature. Currently, you can already jump to the next pattern temporarily by setting the _next pattern_ fill mode and engaging fill. If fill was continuous instead of binary, it could morph to the next track. The other modes could also be more interesting with a continuous fill level. I would then implement _Fill Amount_ as a routable target (for CV or MIDI control) and the fill keys on the performer page would still set _Fill Amount_ to 100% as in the current implementation, so that behaviour would not change.

toneburst, 2019-05-29:
> I think this is an interesting idea. But I would definitely implement it as morphing from one pattern to the next on the same track. Otherwise its becoming a mess.

This was what immediately sprang to my mind, too.

> One possible implementation would be to extend the Fill feature.

That makes perfect sense!

sneak-thief, 2019-05-29:
Sure, using the next pattern actually a very interesting alternative in terms of transitions. In fact, transitions are one of the most important aspects of sequencing and yet the most ignored. 

Fantastic!

toneburst, 2019-05-30:
There are various options in terms of how the morph works.

1. Straight linear interpolation between values, snapping to nearest scale values for notes. This will work reasonably well for notes, gate lengths, and threshold values for per-step randomisation or gate/repeat/note, but won't work for binary values (gates on/off values, slide on/off)
2. Assign random value to each step and use morph-position as a threshold, such that all values of a given step snap from pattern 1 (p1) to pattern 2 (p2) when morph-position is greater than the random value. This way, at morph-position 0, all steps will come from p1, and at morph-position Max,  all steps will come from p2, with intermediate positions causing some steps to be chosen from p1 and some from p2

toneburst, 2019-05-30:
A variation on option 2 above would be to make the random number assigned to each step represent a probability that the step from p2 will play, as opposed to the step from p1.

The morph-position value will apply a bias to the probabilities, much like the routable bias parameters.

sneak-thief, 2019-05-30:
>     threshold values for per-step randomisation or gate/repeat/note, but won't work for binary values 
>     (gates on/off values, slide on/off)

That's precisely why I mentioned using a "probabilistic method". The devil is in the details of course - I left it specifically vague to stimulate Simon's imagination :D

toneburst, 2019-05-30:
I spend a while trying to develop a browser-based (WebMIDI) acid bassline-generator, a couple of years ago, based largely on morphing concepts, inspired by Mutable Instruments Grids Euro module, so I have a few ideas on this ;)

toneburst, 2019-05-31:
> The other modes could also be more interesting with a continuous fill amount

True. Again, this could be done using probabilistic methods.

sneak-thief, 2019-06-01:
Some keywords here: stochastic interpolation, brownian motion.

Check out this for ideas and inspiration:

_Algorithmic Composition Using Probabilistic Methods_

https://quod.lib.umich.edu/s/spobooks/bbv9810.0001.001/1:18/--algorithmic-composition-a-gentle-introduction-to-music?rgn=div1;view=fulltext

westlicht, 2019-07-05:
Will be part of the 0.1.22 release. The most consistent concept was to go with my initial idea and it's actually quite nice and fun to play with!

> One possible implementation would be to extend the Fill feature. Currently, you can already jump to the next pattern temporarily by setting the Next Pattern fill mode and engaging fill. If fill was continuous instead of binary, it could morph to the next pattern. The other modes could also be more interesting with a continuous fill amount. I would then implement Fill Amount as a routable target (for CV or MIDI control) and the fill keys on the performer page would still set Fill Amount to 100% as in the current implementation, so that behaviour would not change.

closed 2019-07-05 by westlicht