westlicht/performer #66: Re-Implement Launchpad Pad Brightness Control

toneburst, 2019-05-29:
I have 2 Launchpad Minis (both Mk.2 models), connected to 2 PER|FORMER modules.

There seems to be a noticeable brightness difference between the two Launchpads, and the brighter unit intermittently looses communication with the PER|FORMER it is connected to.

I can't be sure that the higher LED brightness of my 2nd LP Mini is the reason it occasionally disconnects, but if (as seems to be the case), there is significant variation in brightness between Launchpad units, would it be worth re-implementing the brightness-adjustment setting from 0.1.11?

toneburst, 2019-05-29:
I can't seem to edit the first post, annoyingly. I meant to say my Launchpads are both Mk.1 units.

toneburst, 2019-05-29:
Also, the page created a duplicate Issue when I tried to edit this issue. I've closed the issue (#67), but can't seem to delete it.

Sorry about that. I'm using an unsupported Safari version on my old iMac, so perhaps that's the reason (though I haven't seem these issues before, on the same system).

sneak-thief, 2019-05-29:
The brighter launchpad is likely in "Full Power" mode and not "Low Power". When the launchpad first turns on, it will respectively display a letter F or L depending on the mode it's in:

https://customer.novationmusic.com/sites/customer/files/novation/downloads/6961/launchpad-s-and-mini-advanced-features-guide_0.pdf

Novation recommends using a powered USB hub for the Full Power mode.

toneburst, 2019-05-29:
Ah.. that was it! I've set it to Low Power mode, and so far, so good.

Thank you, Michel!

toneburst, 2019-05-29:
Before I close the issue, could I recommend that the Launchpad Low Power mode is documented in the manual, @westlicht?

westlicht, 2019-05-29:
@sneak-thief good catch, I remember reading about that a long time ago.
@toneburst good idea to mention this in the manual!

westlicht, 2019-05-29:
Manual is updated. Closing this.

closed 2019-05-29 by westlicht