westlicht/performer #69: Audible high-frequency buzz from PSU using Launchpad

toneburst, 2019-05-30:
Both my PER|FORMER modules produce a clearly-audible high-frequency buzzing noise when used with the Launchpad. The noise disappears when the Launchpad is disconnected.

I'm currently powering my 2x PER|FORMER modules from a Mutable Instruments Module Tester, which may be relevant.

westlicht, 2019-05-30:
The high pitched noise is coming from the module or from the tester? In general, the tester is probably not suitable to power the module with a launchpad connected. From the tester manual:

> Because the tester uses half-wave rectification, and input capacitors and heatsinks of moderate size, it is not recommended to draw more than 200mA from the supply rails.

The module takes 100mA without USB power. Leaving 100mA at 12V which will give you around 200mA after DC/DC conversion on the 5V rail. So in essence the DC/DC converter might just struggle to get enough current.

toneburst, 2019-05-30:
It's hard to say where the noise comes from, since the Module Tester and PER|FORMER are close together.

It does sound likely that the Module Tester is struggling to provide sufficient power, though everything does appears to work fine, apart from the noise.

I'll close this.

closed 2019-05-30 by toneburst