westlicht/performer #71: Change pattern with midi cc or note

luisonly, 2019-05-31:
This way you dont have to go into the menu, you can sequence pattern changes and most important of all: I can change patterns for several performers at once by doubling the cc signal and sending it to both performers.

kurtjoy, 2019-06-03:
Was just about to suggest this as well. Thanks

toneburst, 2019-06-04:
Or MIDI program-change?

dijfsanders, 2019-07-01:
Yes! This would be a great feature! It was my first thought on pattern changes

westlicht, 2019-07-01:
Not exactly sure what a good implementation would look like. I guess there are two options:

1. Implement pattern change as a routing target. So one could use MIDI or CV inputs (or even an internal track for that matter) to program pattern changes. Not sure how practical it would be but it would for sure make a nice feature for more experimental stuff.

2. Implement pattern change as part of a "MIDI implementation". This could be a program change message, or some CC message. But essentially it would be a fixed message that the sequencer always listens to. This is a bit against the current design, where MIDI messages are consumed more dynamically by different means (note recording, routing parameters, clock etc.).

luisonly, 2019-07-01:
I personally think number 1 gives greater flexibility... and the program change can be later used for choosing project files via program change midi message.

kurtjoy, 2019-07-01:
Number 1 would be perfect

westlicht, 2019-07-02:
Will be done as part of #89.

closed 2019-07-02 by westlicht

soffer, 2022-12-18:
Late for the party: Is it possible to change patterns with midi messages? And if so, how? Went over the manual and couldn't find it