westlicht/performer #72: Feature request: Record CV and reproduce it as a curve

luisonly, 2019-05-31:
Recording cv movements through the cv in or by midi cc and outputing it as a curve would be a great feature.

sneak-thief, 2019-06-01:
My guess is that there isn't enough memory to do anything like this.

westlicht, 2019-06-03:
No, there should not be a memory issue with this. The bigger issue is how to configure which CV input or CC to use for recording. But I agree it would be a nice feature.

luisonly, 2019-06-03:
My guess would be under "Project" menu, after "CV/Gate input" selection, a "CV record input" and "CC record input" selection options which by default are off.

toneburst, 2019-06-03:
What would be the resolution of the recorded CV?

westlicht, 2019-06-03:
The trick is to match the incoming CV to the best matching curve, or subset of curve types. So I would probably have something like 16 samples that are then used to determine the best matching curve shape. It will not record perfect curves (same as notes are not recorded perfectly) because there is some data reduction taking place obviously. But it might still be a nice feature to quickly record modulation.

toneburst, 2019-06-03:
Ah, so it wouldn't be a realtime recorder, as such. That makes sense.

stansanders, 2019-06-15:
best matching curve would be awesome, especially if it also fits the min/max of the selected curve to try and fit the input signal as well.

westlicht, 2019-06-15:
> best matching curve would be awesome, especially if it also fits the min/max of the selected curve to try and fit the input signal as well.

that's exactly what I do in my experimental branch. Will probably release it for with the next version.

westlicht, 2019-06-21:
Will be part of 0.1.21 release.

closed 2019-06-21 by westlicht

sneak-thief, 2019-06-22:
Looking forward to testing this out with a variety of CV sources! I don't expect a perfect reproduction but it will be an excellent creative tool, like the slew from an envelope follower.

sneak-thief, 2019-06-27:
I just tested this new feature: it functions relatively well when sending CV generated with manual knob wiggles. Sending an LFO at various speeds gives "creative" results (as expected). 

Well done.

luisonly, 2019-07-01:
This is amazing. Thank you very very much. Works as expected :)

commit 124d6354 (2019-06-21): implemented recording curves from cv

fixes #72