westlicht/performer #73: Feature request: configurable encoder direction

patrickod, 2019-06-01:
I just finished a build of my first per|former for which I used an encoder I had left over from a previous project but which I did not realize worked in reverse order to the current one listed in the BOM.

For the moment I plan to make a private build with https://github.com/westlicht/performer/commit/8d133dd566a7de1f454c86306f0f3067abcdde87 reverted to fix this, but I think this would be a great thing to allow users to configure in the systems page similar to the way that the Temps Utile and Ornament and Crime modules do.

patrickod, 2019-06-01:
:man_facepalming: Discovered the `hwconfig_reversed_encoder.hex` files in the repository as I was building this version and have flashed that to fix my own build. Happy to close this out :smile:

westlicht, 2019-06-03:
Yep, with the `hwconfig` you only have to set it once and never care again. Closing this.

closed 2019-06-03 by westlicht