westlicht/performer #74: Flip LED colors as hwconfig option

westlicht, 2019-06-07:


sneak-thief, 2019-06-08:
This is useful! Thanks.

luisonly, 2019-06-10:
Very useful for us colorblind. Green and red leds look the same.

sneak-thief, 2019-06-10:
FWIW, if you're colour blind, high-brightness Red/Blue leds also work: I recommend 5k resistors for the blue leds ( R76, R74, R72 & R70) and 2k resistors for the red leds  (R77, R75, R73 & R71). This is presuming the red pin of the red/blue led is on the same side as the originally specified red pin of the red/green leds.

[image]

luisonly, 2019-06-10:
Amazing! Thank you, but theres 32 LEDS and you only tell me to change 8 resistors...

> FWIW, if you're colour blind, high-brightness Red/Blue leds also work: I recommend 5k resistors for the blue leds ( R76, R74, R72 & R70) and 2k resistors for the red leds (R77, R75, R73 & R71). This [...]

sneak-thief, 2019-06-10:
It's the magic of multiplexing!

One 74HC595 shift-register handles the rows of buttons/leds (U12) and the other handles the 4 columns of leds (U13). However, because the leds are bicolour, there are 4 shift-register outputs for the red labelled RCOL0-3 and 4 for the green (or blue) labelled GCOL0-3:

[image]

luisonly, 2019-06-10:
whoa! thanks man, I appreciate it.

westlicht, 2019-06-12:
As mentioned on the build page, please make sure to not sink too much current on the shift registers when changing the leds or you might damage the chips.

The led invert option will be available with the 0.1.20 release coming up.

closed 2019-06-12 by westlicht

sneak-thief, 2019-06-13:
I measured the DC current draw of a few different high-brightness red/blue LEDs on a breadboard: when set to a similar brightness as on my Performer, the blue used 0.3mA and the red was 0.4mA.

A typical 74HC595 can easily deliver 35mA (it's max rating) and likely twice as much if you want to walk on the wild side: https://electronics.stackexchange.com/questions/286055/how-is-the-curr ent-limited-through-the-leds-in-this-led-matrix-design

Even with all 32x red/blue LEDs on, my estimate is that it's under 30mA. This is ignoring the fact that the Performer is also driving them with PWM, so even less current.