westlicht/performer #75: Feature request: additive sequences

stansanders, 2019-06-15:
###Motivation:
Create new patterns using the data that already exists in other patterns for exciting performances and rapid improvisation.  

###Idea:
Combine two sequences of one track into a new combined output.  The simplest iteration of this would be to combine the gates of the two sequences as well as the assigned CV from each, adding the CV values together for any notes that have an overlapping gate.  If the combined patterns are different lengths and not evenly divisible this creates the potential for the new combined output to have a very long set of variations as the two sequences drift in and out of alignment.

##Further extensions:
#Additional math:
  This function could be extended to other math functions like subtracting one sequence from another (the second sequence is subtracted from the first) playing only the union of the two sequences (only the overlapping notes are played, with their CV values adding to each other), or playing only the differences. 
#Combine many:
 Also could allow combining more than two sequences.
#Octave combine:
Each semitone of the first sequence shifts the second sequence by 1 octave.
#Gate roll:
The gates of the second sequence create a roll if they overlap with a gate on the first sequence.

###Interface:
##Playing pattern combinations:
In pattern / clip launch view on the launchpad, hold down the button corresponding to one sequence, then while still holding that button down press the button corresponding to another sequence on the same track.  Both buttons light up red to indicate they are in combined mode (compared to usual selected clip being green).  Unlatch this mode by pressing either of the latched sequences indicated by red buttons and the track will return to normal single pattern play mode.

##Pattern combine settings:
In track settings, select sequence gate combine mode (add, subtract, union, difference, octave) and cv combine mode (add, subtract, union, difference, roll).

###Caveats:
This might be less ergonomic using the performer buttons compared to the launchpad.  I don't know if the system will have a hard time with iterating multiple sequences at a time, or adding values before outputting them.

westlicht, 2019-06-15:
Not to fond of that idea, as it is a very specific feature request that most people probably won't use. Also, you have a tremendous amount of possibilities already using routings to have multiple sequences affect each other. Sure, it's probably not the most immediate to use in a live situation, but for that I would suggest to use external modules to combine multiple sequences which might be much more hands on. Also, there is the _snapshot system_ which allows for quickly making variations of patterns.

stansanders, 2019-06-15:
I think you might find people would use it quite a bit actually as the functionality is related to a popular technique for live play.  People like steevio and mylarmelodies are big proponents of combining gates and using precision adders to combine CV sequences and they preach this to a wide audience.  To accomplish even the most basic version of this requires  more than one external module, and doing stuff like union / difference would be even trickier externally.  

Without committing to implementing it, can you comment on how feasible the proposed functionality and interface are under the current data model?

westlicht, 2019-06-16:
> I think you might find people would use it quite a bit actually as the functionality is related to a popular technique for live play. People like steevio and mylarmelodies are big proponents of [...]

Yes, I also use external gate combiners in my setup, it's nice to have some hardwired switches for performance. I use this mostly for drums to allow switching between and combining sequenced patterns and more procedural sources like euclidean patterns.

> To accomplish even the most basic version of this requires more than one external module, and doing stuff like union / difference would be even trickier externally.

True, but I also think that the magic of modular is being able to combine simpler blocks into more complex things. If you want all this flexibility, the ultimate setup is using a laptop, software and some controllers. However, we all in part use modulars to escape from this, isn't it? I think that my sequencer is already getting to the point where it becomes too complex to be considered simple and intuitive to use and I strongly want to avoid putting in too many new features that are not well integrated with the rest of the system.

> Without committing to implementing it, can you comment on how feasible the proposed functionality and interface are under the current data model?

Implementing a system as you propose it will conflict in quite a few places with the existing design. Just to name a few without thinking too much about it:
- Currently each track is able to play back one sequence. There is quite some bookkeeping to do for stepping through a sequence and generating the output signals. All of that would have to be multiplied to run multiple sequences per track to allow combining them.
- How would recording work if multiple sequences are playing?
- The mutes/solos and fills will be a bit weird if multiple sequences are playing.
- Taking a snapshot of a pattern is a bit weird if there is a combination of sequences running.
- Controlling all of that from the existing UI is more than tricky IMO.

All of that being said, the software is open source and people that want to evolve it into their own direction are very welcome to do so. Having a prototype that I can test would help a lot to assess the potential of an idea and might push me towards adapting some new features. For me the main goal is to make all of the functionality more consistent and try to add features that blend well with the existing system. There is still a lot of things on my list which are way more straight forward and still quite challenging to do in a nice way. I hope that explains my initial reaction that I'm not too fond of the general idea.

stansanders, 2019-06-16:
I see where you're coming from, also appreciate the fact that you've got the entire system to consider where I'm not burdened with development for the entire module.  I may try implementing this feature on my own branch and if successful I agree it'd allow this discussion to proceed further.  Thanks for the detailed response.

closed 2019-06-16 by westlicht