westlicht/performer #76: Feature request: new gates default to match note of the most recent note in sequence

stansanders, 2019-06-15:
### Motivation:
When programming a note sequence, I tend to start out by making a slow pattern to establish an overall sequence.  I then add more notes into the empty spaces to fill in the pattern.  Currently, when starting from a blank pattern in this method, when first adding a note it defaults to a value of 1+0 which makes it hard to fill patterns in while playing live without having moments that sound bad as I frantically dial them up to fit.

### Idea:
If a note has a value of 1+0 and no gate set (so we assume this note has never been set before), but then a gate is added on that note, change its note to be the same as its neighbor to the left.  This means that if you punch in a sequence of notes on steps 1, 9, 17, and 25 and program their notes, then tap in gates on steps 18,19,20, the notes of steps 18, 19, and 20 will immediately set to the same note as step 17.  If you then remove the gates on 18, 19, and 20 their notes will be left alone, and adding gates back onto those steps will not affect the notes.  The auto-setting behavior can be brought back either by clearing the pattern or manually setting the notes to 1+0.

This will allow quick creation of patterns by setting the overall progression quickly with a few steps, and then adding additional notes that default to playing along with the existing pattern and can also be edited more quickly and musically as they start out as part of the existing sequence.

### Outside example / further explanation:

I currently accomplish a similar functionality in one of my racks by having the gate sequencer (circadian rhythms) controlling envelopes separately from the pitch sequencer (ER-101).  By doing this, I can freely add and remove gates and their notes always follow the existing sequence.  I then proceed to add notes inside the sequence by using the ER-101 note split function.  I think that part of the reason I really like this is because it allows changing the rhythm and feel of a sequence in a live setting without messing up the notes or having to devote two hands to the task.  I think adding the functionality outlined above would make improvisation and live programming with the performer truly effortless, so you could do things like set notes on steps 1 and 33 to establish a long base pattern and start improvising notes into it without interrupting the good flow of things.

westlicht, 2019-06-15:
I do like this idea, the only problem is how to safely detect that a step's note value has not been edited before. This might lead to some weird behaviour in edge cases which I don't like.

stansanders, 2019-06-15:
I'd argue that testing for the 1+0 note value is pretty fair, but I can understand your point.  I would also say we don't want to place a flag that has to be cleared by clearing the pattern or initializing the project because it'd make it harder if you want to bring back the follow behavior while editing and keeping your pattern.  

What about having this feature disabled by default and enabling it from a config menu, at least while its using my proposed 1+0 as the test for whether to follow?  Maybe worth trying out the simplest implementation of this idea before worrying too much about edge cases.

westlicht, 2019-07-05:
I have given this quite a bit of thought and was not able to come up with a nice consistent concept for implementing this. I don't like to have an _option_ to enable this behaviour either, makes things more convoluted than necessary. I have thought of alternate ways to achieve the same effect but didn't really get to a good solution. So for these reasons this will have to go to the backlog. @stansanders hope you understand and feel free to brainstorm more!

closed 2019-07-05 by westlicht