westlicht/performer #77: Feature request: gate paint / bleed for making long multi-step gates

stansanders, 2019-06-15:
### Motivation:
I don't see an easy or quick way to blast long gates into a sequence, so if you need to create a note that is 16 steps long you must add their gates manually, dial up their lengths manually, and program in their notes manually.

### Idea:
Add an option I would call "gate painting" or "gate bleeding" where a gate's length can be adjusted to be longer than its step in order to paint/bleed further into the sequence. By adding a bleed/paint function you could quickly create long multi-step notes which are easy to edit live as they can be edited from the single origin note rather than manually editing every step.

### Interface:
In gate length view, begin adjusting the length of a gate on one step.  Continue increasing the length until it bleeds into subsequent gates.  These steps will now illuminate to indicate they are active and cannot be edited.  The gate length indicator for paint/bleed steps will change to a (slash, dashed line, or some other indicator) to show that they are part of a painted/bled note and they will no longer be editable except by editing the note that is bleeding/painting onto that step.  The CV values of notes within a bled/painted region will follow the CV values of their respective origin note.

**Additional function:**
To rasterize (to borrow a term from photoshop) bleed/paint data onto a note so that it can be edited individually, press and hold the origin note and then tap the note you wish to rasterize.  This note will no longer be affected by changes to the origin note, and can be edited as normal.  

To re-link a rasterized note to a bleed/paint note, press and hold the origin note and then tap the rasterized note to reunite them.  This note will now behave as part of the existing paint/bleed note.  Notes must be within the bounds of a paint/bleed note in order to be linked back into it.

westlicht, 2019-06-15:
You can edit multiple steps using `SHIFT` and selecting steps first, then go through layers and adjust values. So you can make a 16step long gate simple by holding `SHIFT` and swiping over the 16 steps, then adjust gate length to max.

The problem with having gate length longer than a step duration is that it doesn't match the data model anymore. So I don't really like this idea too much. I have a new track mode in a development branch that might be more useful for that kind of thing (variable step length).

sneak-thief, 2019-06-15:
And double-clicking SHIFT while in STEPS mode selects all 64 steps: 

> Advanced Step Selection

When holding SHIFT and pressing any of the step buttons, steps are selected in a persistent mode. This means that steps are kept selected even when the step buttons are released. This allows to select multiple steps across different sections. Steps can also be removed from the selection by holding SHIFT and pressing step buttons corresponding to selected steps again. To clear the entire selection, simply press any step button without holding SHIFT or quickly press SHIFT twice (double tap). When the selection is empty, double tap SHIFT to select all 64 steps.

Finally, there is a shortcut to select a series of steps. For example, to select steps 1, 5, 9, 13 and so forth, hold SHIFT and S1 and then double tap S5 while holding the other two buttons. With this shortcut, any series of steps with a common interval can be selected quickly, great to build rhythms.

    Note: Different series can be combined by applying the shortcut multiple times. Each time it is executed, the steps of the series are added to the selection.

>

closed 2019-06-16 by westlicht