westlicht/performer #78: Feature Request: Launchpad sequence-segment indicator + "follow mode" during playback

sneak-thief, 2019-06-16:
Issue:
When using the Launchpad in Sequence mode to notes, lengths & retrigs, it is often disorienting because there is no indicator of which 8-step segment of a sequence is currently being played (if the sequence is longer than 8 steps).

Editing notes, lengths & retrigs is also slowed down since you have to press 1 plus the sequence segment to navigate to the corresponding part of the sequence that you wish to edit.

Enhancements:
**1. Current Segment Indicator**
When editing the notes, lengths & retrigs of a sequence, use dimmed green lights on buttons 1-8 to display if that segment has notes. The button number of the active segment would be lit in red. Normally, if a sequence only has 8 or less steps, then only one button would be lit in red as it would always be active. However, there are cases where the first step of an 8-step sequence could start at step 8 and end at step 15, in which case button numbers 1 and 2 would be lit up accordingly.

When using the regular controls (8+7 start/stop, 8+1 pattern mode, 8+2 sequence mode, etc.) the top leds would behave the way they did before, with the corresponding pressed buttons lighting orange as usual.

**2. Segment Follow Mode**
During playback, it would also be helpful to have the option for the currently active 8-step sequence-segment to automatically switch so it always displays the steps that are currently being played. 

This "follow mode" is common with many current hardware step sequencers, eg. NI Maschine.

Suggested UI change:
Holding down Launchpad button number button 6 would display a first row of either red or green leds in the button grid. Each GRID led would represent the follow mode enable/disable status of each track. Holding down button 6 + GRID would toggle between follow mode being enabled/disabled for each track.

Follow-Mode caveats: 
A downside could be if a sequence is 1 or 2 steps longer than the previous page, eg. 9 or 10 steps - or if the first sequence steps were 7 or 8. The corresponding last or first page could possibly change too quickly. 

Also sequences with fast tempos become harder to edit while playing and switching.

FWIW neither of these downsides have prevented this follow mode to appear in other major hardware step sequencers.

westlicht, 2019-06-18:
> 1. Current Segment Indicator

I don't like having the upper leds showing the selected segment, these are the main function keys. I think having to press `1` to see what segment/window is active is fine. While editing, you probably know where you are anyway. Having the ability to use `1`+`2` and `1`+`3` to move left and right is a quick way to navigate left and right.

> 2. Segment Follow Mode

What is the advantage of having that? Just eye candy? Because I don't see how this is useful for editing. Editing gates is available on a single page anyway, so no need for follow mode. Editing the other layers while the sequence is running through the segments seems rather unuseful other than pressing some random buttons and hoping for some happy accidents. Having to add enable/disable per track for this feature also seems way overkill to me.

So please convince me with some examples of how this would be really useful :) but for now I don't see much value in it at all, but adding additional complexity to an already quite crowded interface.

sneak-thief, 2019-06-19:
Re. Current Segment indicator - let me try again to explain:

Picture yourself playing the performer. It's in playback mode and you have all 8 tracks playing. Some of the sequences on those tracks are up to 64 steps long. 

During playback, there are times when you want to actually change the notes, lengths and retrigs of the entire sequence as it plays ("on the fly", as they say). The Launchpad is great for this - except for the fact that you can only edit 8 steps at a time. 

The problem is there's no visual indicator of _which_ sequence-segment is currently playing vs what is currently selected. Currently, a red chaser-led will indicate which of those particular 8 steps is playing but beyond that, when any other sequence-segment is playing it's not necessarily clear which other segment is playing.

Why does this matter? 

My feeling after doing this for 20 years is that performing live is about managing chaos. Many times you just want to jump in and improvise by adding note variations, especially in a performance situation. (Even a TR-808 has indicators for steps 1-16 and steps 17-32 and can automatically switch between the two.)

But your attention is also focused on the synths/samplers as well! So every time you glance back at your sequencer, it's essential to recognize where you're at in a sequence.

Performing live isn't always just a matter of muting/unmuting, changing patterns and turning gates on and off.  Sometimes you want to make changes in anticipation of what is about to happen instead of just focusing on a specific sequence segment.

EXAMPLES:

1. On Maschine Mk1, the display shows which of the current 16 steps have gates - and the bar below indicates which 16 of the 64 steps are active. Any of those notes can be selected and modified during playback.

[image]

Furthermore, the other screen displays the current measure so you always know where you are in the sequence. You can also see the FOLLOW button which automatically switches the sequence segment so you can edit notes as they appear.

2. On Squarp Pyramid, there's a similar line-segment indicator at the very bottom of the screen showing which part of the sequence of playing back.

[image]

3. The Yamaha RS7000 here is indicating the currently edited 16 steps, in this case the 2nd measure out of 16 (MEAS 002/016). 

[image]

(The BPM indicator here has also been switched to instead show the current measure - very useful)

NOTE: The final measure length on the RS7000 & Maschine (eg. 001/**016**, 3:1/**4:4**, etc.) is determined by the longest pattern. If you ever decide so have a measure counter like most other sequencers, then the "total" length is simply that of the longest of any of the patterns.

PS. Here's a basic pattern sequencer for audio loop-stems that I programmed in C in 2006: http://cdm.link/2012/10/sneak-thiefs-sneaquencer-is-a-diy-monster-dream-hardware-for-performance-open-source-music/ - when I perform live, the most important indicators are the ones telling me which pattern is playing and which measure is playing.

sneak-thief, 2019-06-19:
FWIW these other sequencers follow the sequence during playback and change segment views accordingly, allowing for realtime editing of each active segment notes:

Beatstep Pro 
Elektron Octatrack, Digitone, Digitakt, Analog 4 (w/ segment indicators)
Ableton Push 2 (w/ segment indicators)
Toraiz Squid (w/ segment indicators)
Torais SP-16 (w/ segment indicators)

BTW, Ableton Push 2 also has an interesting solution for simultaneous gate and note entry with an 8x8 grid:

https://youtu.be/GVilj3bChHY?t=37

rolfjack, 2020-10-01:
First of, let me say that the performer is such a gem i modular. I just bought a Launchpad mk2 and it adds so much power.

After playing with it for a wile, I must +1 to the idea of follow mode. But perhaps in a simpler implementation.

On my wish list would be the following feature additions:
- Double-press/long-press buttons A-H in order to toggle between follow mode or ordinary mode. This could either be a simple global setting or a per channel setting.

During playback it is super useful that buttons A-H flash when gates are played, and show their muted status.
- Having an extra play mode (for instance 8+6) where the top row of control buttons would flash when gates are executed, and switch position as the sequencer moves through pages for the currently selected track would be immensely useful. In combination with a follow mode feature it would be very obvious what is going on.

Finally (and this is perhaps the most important item for me): IT would be such a book if it was possible to toggling gates on/off in notes mode via a double press or long press on a pad.

When editing a melodic sequence it is super convenient that one can pick 8 notes in scale or semitones. However, it is quite cumbersome to have to switch to gate mode, toggle gate status, return to note mode and set a note in order to make a small change to the rythm. It would be so much more intuitive to simply double tapping a note to change the gate value on/off.

It is worth considering if long pressing one or more notes simultaneously could be used to make a tie or something. But adding some flexibility in how notes and gates can be edited on the notes page would make editing significantly faster and more intuitive.

Thanks for creating an amazing tool.

GeoffMaciolek, 2020-10-28:
I believe there was a miscommunication here; I also misunderstood the "follow mode" request.

I believe the request is not for the Launchpad display to **scroll**, but merely to change the displayed "page" to the currently-playing 8 steps: a single jump every 8 steps.

("Follow mode" in some sequencers - such as trackers - is a constantly-scrolling view, which I agree would be a real mess to use on the Per|former / Launchpad combo! Though it *would* make nice eye candy. Not my priority though! ;)

rolfjack, 2020-10-29:
Yes, that is also my understanding of the most useful follow mode: An automatic page turner so that one can see (and possibly edit on the fly) the active part of the sequence.

It would be a great improvement.

shsCT, 2020-12-22:
Absolute +1 for this feature. 

When playing, this is the number one challenge with using it; let’s say I start a whole track by live recording a 4-bar sequence on a Performer track where the melody doesn’t really cue the “beginning” to my ear.  Then I move about the cabin, adding various ambient parts on other gear in my live kit. Now I want to start recording on the “1” in the Performer to properly land a start of the “1” in the track. If I look back to the performer, it’s not at all quick to determine where the performer is in it’s 64 step loop to prepare me for recording a new Performer track on the downbeat. The best at this point is to wait and watch knowing which page I’m on - but this often means I have to then wait at least two more passes to jump in with the new track - which adds a lot of “hang” time in the music performance. Because of the way I like to create music, this is something that happens in almost every track. Even four blinking lights somewhere on the interface (similar to the way Elektron handles this with their page lights) would help immensely.

I love the Performer so much that I own three of them. It is without a doubt the most fluid sequencer to work with in eurorack!