westlicht/performer #81: Feature request: Tracks in curve mode retain the ability to also output a gate sequence

stansanders, 2019-06-18:
Not sure if the model is such that this isn't possible, but it seems like it'd be handy to keep the gates functionality when a track is in curve mode.  Is the current state a result of some prohibitive factor?  Seems like the curve track having the same gate UI as a note track, but with the curve UI replacing the note UI would be ideal.  I'm finding myself having to decide between adding a secondary gate sequencer or forgoing curve tracks.  Having gate functionality exist on a curve track would mean curve tracks could be sequenced while also using those channels' gates to handle drum sequencing.  I don't see any reason the curve and gate layers would have to interact, it'd essentially be giving a free gate-only sequence anywhere a channel's CV is being used for curves.

toneburst, 2019-06-20:
I suggested a while back a new "dual trigger" track mode that would use both CV and gate outputs as twin triggers, and have two independent trigger layers, instead of one CV/Note and 1 gate.

Simon wasn't keen, but I still think it's a good idea.

westlicht, 2019-07-22:
Will be part of 0.1.27, but it's not going to be a copy of the note track behaviour.

closed 2019-07-22 by westlicht

toneburst, 2019-07-25:
That’s cool. I speak a little Binary, so it makes sense to me, but might be a little unintuitive to some.