westlicht/performer #82: HWconfig options in System/utilities

kurtjoy, 2019-06-30:
Add software switches for LED colour swap and reverse encoded (especially) for those with STM32 programmers. Somewhere in the system menu wopuld be an ideal location.

westlicht, 2019-06-30:
I didn't want to expose those in the menu because this really should be programmed once and then never again. Also, this module is DIY, so you probably do have a programmer if you built it.

closed 2019-07-01 by westlicht

toneburst, 2019-07-27:
Does applying an update via the SD card overwrite the system settings for encoder direction and LED polarity?

westlicht, 2019-07-27:
> Does applying an update via the SD card overwrite the system settings for encoder direction and LED polarity?

No, the whole idea of the hardware config is to write it **once** and never care about it again. This way I can distribute a single update file that everyone can use. The firmware has no hardcoded configuration, it is either using defaults or the settings written with the hardware config.

toneburst, 2019-07-27:
Good to know.

I think the LED polarity is something that a user might want to change as an aesthetic choice, so it might make sense to add this to a settings page.

I agree they're unlikely to want to change encoder direction, unless they've had to replace a faulty encoder with a different part.

westlicht, 2019-07-27:
The led polarity is probably a one time setting too, as it does not really make sense to switch it. I mainly added it for people that manage to solder in all the leds reversed :) Again, if you switch hardware you have to reflash the hwconfig once. I agree that for DIY people it might make sense to have this in a menu somewhere, but this is merely for their convenience. Reflashing is really easy if you built the unit yourself. It's just much cleaner to not have this exposed in the application.

kurtjoy, 2019-07-28:
is there a way to reduce the brightness of the LEDs without having to
change resistors...Mine are blinding!!