westlicht/performer #83: Feature request, previous gate links to next like Eloquencer.

Pipstapee, 2019-06-30:
Not sure if anyone has requested similar.
On the Eloquencer you can hold, say step 1 gate down and it links to the next gate say step 4. Locking the gate high between step 1 and step 4.
Would that be of benefit at all?
I've just started using the Per|former from using an Eloquencer so I'm not 100% of all of it's current features.

westlicht, 2019-06-30:
You can do the same by setting the step length to the maximum value, which "links" or ties two steps together. If step 1-3 are all at max length then you essentially have steps 1-4 as a single gate. You can still have different notes on these steps if you want.

Pipstapee, 2019-06-30:
That makes sense. I hadn't thought of that, I think I've got to untrain
myself from the way the eloquencer works and your manual from front to back
and make no assumptions.
Thanks Simon for taking the time to respond, your sequencer is dense with
modulation features that I think it's easy to overlook them.
Two updates within 2 weeks, when do you sleep?
Thanks again for taking the time to explain something that's bound to be in
the manual.

closed 2019-06-30 by westlicht