westlicht/performer #89: New routing targets

westlicht, 2019-07-02:
Add new routing targets:
- track mute
- track fill
- track fill amount
- pattern

westlicht, 2019-07-02:
Fixes #51, #65 and #71.

sneak-thief, 2019-07-03:
If what I'm envisioning is at all possible, I'm very curious to test some recursive muting scenarios with several tracks un/muting one another. This is Cirklon territory :D

westlicht, 2019-07-03:
> If what I'm envisioning is at all possible, I'm very curious to test some recursive muting scenarios with several tracks un/muting one another. This is Cirklon territory :D

Haha, sure, why not :D

westlicht, 2019-07-05:
To be released with 0.1.22.

closed 2019-07-05 by westlicht

dijfsanders, 2019-07-07:
The Pattern change over midi seems to be buggy. If i use more than one pattern routing, it doesn't respond anymore. It seems to react badly on Note numbers instead of CC. Only if i route pattern [1-16] to a CC, it respond correctly. (i use a AKAI APCKey25)

westlicht, 2019-07-07:
> The Pattern change over midi seems to be buggy. If i use more than one pattern routing, it doesn't respond anymore. It seems to react badly on Note numbers instead of CC. Only if i route pattern [1-16] to a CC, it respond correctly. (i use a AKAI APCKey25)

That's expected behaviour. The implementation does not allow to have multiple routes targeting the same target parameter. Only if you target different tracks, you can have multiple routes going to the "same" target (but on different tracks). I think there are two things that could improve this:

1. Do not allow setting up conflicting routes in the first place.
2. Add a new "note range" source, that routes a range of MIDI note values to some target value. E.g. you could setup MIDI notes C3-D#4 as a range to control the pattern number.

dijfsanders, 2019-07-08:
Thanks for the feedback!
So, this 'Note Range' is something you will implement in a next update?

greets
D

westlicht, 2019-07-08:
Yes, prototype to address both points is already working.

commit e28de746 (2019-07-05): added playstate routing targets (mute, fill, fill amount and pattern)

fixes #89