westlicht/performer #90: Launchpad Pro support please

EL-LYN, 2019-07-03:


sneak-thief, 2019-07-03:
What's the exact problem that you're having? It should be working: 

> The following Lauchpad controllers have been tested and are working:
>
>   Launchpad Mini MK1
>    Launchpad Mini MK2
>    Launchpad S
>    Launchpad Mk2
>    **Launchpad Pro**

- https://westlicht.github.io/performer/manual/

westlicht, 2019-07-03:
@EL-LYN have you connected the Launchpad Pro via USB? Controllers only work via USB.

westlicht, 2019-07-06:
Closing this for lack of answer. @EL-LYN feel free to report back if the problem still exists.

closed 2019-07-06 by westlicht

Klangkontrolle, 2019-07-20:
Deutsch: Hallo ich hätte eine Frage. Im Note Modus zeigt das Launchpad nur 8 Steps an. Wie kann ich zu den restlichen Steps wechseln wenn ich mit 64 Steps arbeiten möchte? Danke
Englisch: Hello, I have a question. In the Note mode, the Launchpad displays only 8 steps. How can I switch to the remaining steps if I want to work with 64 steps? Thank you

westlicht, 2019-07-20:
@Klangkontrolle please read the manual, there is a section on navigation. From the manual:

Because the 8x8 grid of the Launchpad can only represent part of the data to be edited, the grid only acts as a window into a larger virtual 64x64 grid. Hold `1` to enter navigation mode, where the position of the window can be selected:

| Buttons | Description |
| :--- | :--- |
| `1` + `2` | Move window left |
| `1` + `3` | Move window right |
| `1` + `4` | Move window up |
| `1` + `5` | Move window down |
| `1` + `GRID` | Move window to specific position |