westlicht/performer #97: Euclidean bugs

luisonly, 2019-07-16:
Hi :) Loving everything about the seq, and the euclidean generator is amazing for polyrhytms and stuff. Now the problems I have encountered:

1. When I apply a euclidean generator to a track, I lose the routing (in this case midi routing) I had assigned for start and end points in the sequence.

2. If the track is playing 1/32 (or any other division) euclidean generator will change it to 1/16... itd be nice that it sticks to whatever division the track was set to. 

Thank you simon!

westlicht, 2019-07-17:
> 1. When I apply a euclidean generator to a track, I lose the routing (in this case midi routing) I had assigned for start and end points in the sequence.

This is caused by the same issue as #98.

> 2. If the track is playing 1/32 (or any other division) euclidean generator will change it to 1/16... itd be nice that it sticks to whatever division the track was set to.

Good catch, this is because using the euclidean generator actually clears the pattern and resets it to default values. Will change this.

westlicht, 2019-07-18:
Will be fixed with 0.1.25.

commit 59e2d8ec (2019-07-18): made euclidean generator non-destructive

- do not clear pattern but only edit the selected layer
- generate multiples of the euclidean pattern to fill up all 64 steps

fixes #97
closed 2019-07-18 by westlicht