westlicht/performer #99: Launchpad mini issue

luisonly, 2019-07-17:
Hi simon, I just bought a used launchpad mini mk1 and cant seem to get it to work as a control surface. It is recognized as a midi device and mainly just sends notes. Is there something that I might be lacking here like updating the launchpad firmware or something like that? thank you!

westlicht, 2019-07-17:
Can you check the usb vid/pid of the launchpad!?

luisonly, 2019-07-17:
I get: USB\VID_1235&PID_0043&REV_0000

westlicht, 2019-07-17:
Can you retry after setting device id to 1 according to https://d2xhy469pqj8rc.cloudfront.net/sites/default/files/novation/downloads/6961/launchpad-s-and-mini-advanced-features-guide_0.pdf

westlicht, 2019-07-17:
I just tried myself and changing the ID actually changes the USB PID which is used by performer to detect the connected device. So, to have your launchpad detected as a controller it needs to be set to ID 1. I will have to add a note in the manual and maybe it's soon time for an FAQ :)

luisonly, 2019-07-17:
Amazing! it is working flawlessly now! Thank you Simon!

westlicht, 2019-07-18:
Updated user manual.

closed 2019-07-18 by westlicht