westlicht/performer #103: adjust max and min curves at the same time

madeinspace, 2019-07-20:
It would be nice to adjust the curve offset for both min and max value at the same time to keep the range unmodified. thanks for considering! :)

westlicht, 2019-07-20:
Do you have a specific idea of how this would be accessible? SHIFT is already used for fine adjustment.

soporus, 2019-07-20:
Maybe an offset mode added to F4?

westlicht, 2019-07-20:
@soporus I consider that a break from the current design because the function buttons are used to select layers. Each layer represents a piece of per-step data and is represented exactly once. Adding an offset button would imply there is another layer, but that is not the case as it would change the existing min/max layers. In addition, these additional function buttons will probably be used in the near future for other layers that I'm considering to add to curve sequences. So I'm afraid I'm not too fond of that idea.

westlicht, 2019-07-20:
Just to be clear, I think there would be some value in being able to "offset" curve segments, I'm just not sure what the best way of achieving this could look like.

madeinspace, 2019-07-21:
@westlicht I'll have a think, maybe by holding max and min while turning the encoder?

westlicht, 2019-07-21:
> maybe by holding max and min while turning the encoder?

had the same thought, but it also doesn't fit too well with the existing UI

westlicht, 2019-07-21:
I've just implemented a version where you can keep holding min or max (not and) function keys to offset the curve segment. I'm not sure I like this yet, but it's an option.

westlicht, 2019-07-22:
Will be part of the 0.1.27 release.

commit 61760668 (2019-07-23): allow curve min/max to be adjusted at the same time

fixes #103
commit 8a73a08b (2019-07-24): allow curve min/max to be adjusted at the same time

fixes #103
closed 2019-07-22 by westlicht